zachjs/sv2v #206: illegal statements with interfaces which do not either raise an error, or generate valid verilog

hughperkins, 2022-05-08:
Consider the following:
```
module foo();
    ReadPort read_port();
endmodule
```
This runs without any error, outputing:
```
module foo;
	ReadPort read_port();
endmodule
```

I suppose ReadPort could be imagined to be a verilog typedef, but I think that if such a typdef hasn't been defined, then this code should be illegal, and should raise an error, when compiled with sv2v?

```
interface ReadPort;
    logic [3:0] addr;
endinterface

module foo();
    ReadPort read_port();
    ReadPort n_read_port();

    initial begin
        read_port <= n_read_port;
    end
endmodule
```

This generates:
```
module foo;
	generate
		if (1) begin : read_port
			wire [3:0] addr;
		end
		if (1) begin : n_read_port
			wire [3:0] addr;
		end
	endgenerate
	initial read_port <= n_read_port;
endmodule
```
As far as I know, assigning one interface instance to another is illegal? In any case, the generated verilog is certainly illegal. I feel that the system verilog above should raise an error, from sv2v, when compiled with sv2v.

hughperkins, 2022-05-08:
Another example. Add a task to an interface that uses that same interface in the ports, like e.g.:
```
interface ReadPort;
    logic [3:0] addr;
    task copyFrom(ReadPort dst, ReadPort src);
        dst.addr <= src.addr;
    endtask
endinterface

module foo();
    ReadPort foo();
endmodule

```
This compiles with sv2v to:
```
module foo;
	generate
		if (1) begin : foo
			wire [3:0] addr;
			task copyFrom;
				input ReadPort dst;
				input ReadPort src;
				dst.addr <= src.addr;
			endtask
		end
	endgenerate
endmodule
```
However, this is not legal verilog becuase of the ReadPort usage in the copyFrom task. Therefore the sv2v step should fail I feel.

hughperkins, 2022-05-09:
(basically, if something is declared as an interface, and then still appears in the output verilog, then that should likely automatically trigger an sv2v error perhaps?)

hughperkins, 2022-05-09:
(I feel like the goal should be that *all* compile errors are captured by sv2v, so that the verilog compilation *always* succeeds, see https://github.com/zachjs/sv2v/issues/194 )

zachjs, 2023-05-07:
I've added checks to handle the example you gave in https://github.com/zachjs/sv2v/issues/206#issuecomment-1120475769, which now results in: `sv2v: declaration dst uses interface name ReadPort where a type name is expected, within scope foo.foo.copyFrom (use -v to get approximate source location)`.

I haven't yet handled the example you provided in https://github.com/zachjs/sv2v/issues/206#issue-1228968076.