zachjs/sv2v #207: signed to unsigned conversion in struct members causes trouble with >>>

siquus, 2022-06-03:
Hi,
thanks a lot for the tool! Here's the problem I experienced:
System Verilog:

```
typedef struct packed {
		logic [10:0] Exp;
		logic signed [53:0] Mant;
	} normalSigned_t;

normalSigned_t double;
logic signed [54:0] shiftedLargerMant;

always_comb shiftedLargerMant = {double.Mant[53], double.Mant >>> Shift};
```

sv2v changes the assignment to
`always_comb shiftedLargerMant = {double[53], double[53-:54] >>> Shift};`
That is the arithmetic shift will now interpret the left operand as unsigned and work the same as regular shift.

My quickfix for now is introduce an unnecessary `$signed()` to the original System Verilog
`always_comb shiftedLargerMant = {double.Mant[53], $signed(double.Mant) >>> Shift};`

Thanks!

zachjs, 2022-06-19:
This is a great catch. Thank you for filing this issue! I've just pushed a fix. Can you confirm it's working for you?

siquus, 2022-06-28:
That fix looks familiar :-)

Works on my desk - kudos for your fast reaction time!

closed 2022-06-28 by siquus