zachjs/sv2v #209: Parse error: missing expected `endfunction`

hello-eternity, 2022-07-04:
hello ,my sv file have such part
```
function [ROB_PC_OFFSET_WIDTH-1:0] get_pc_offset;
        input [VADDR_WIDTH-1:0] pc_i;
        assign get_pc_offset = pc_i[1 +: ROB_PC_OFFSET_WIDTH];
    endfunction : get_pc_offset
```
and got an error at the third line. here is the file 
[rvh_rcu_pkg.sv.zip](https://github.com/zachjs/sv2v/files/9036632/rvh_rcu_pkg.sv.zip)
command is 
`sv2v rvh_rcu_pkg.sv -w rvh1.v`

zachjs, 2022-07-04:
Although this feature is not in the latest release, the version on master supports procedural continuous assignments. Can you give that a try?

zachjs, 2022-07-11:
Did upgrading fix this issue for you?

hello-eternity, 2022-07-11:
Yes, the master version sovle this problem. Thank you very much.

closed 2022-07-11 by hello-eternity