zachjs/sv2v #211: fail to make

hello-eternity, 2022-07-08:
Here is the error log, How can I sovle this 
```
mkdir -p bin
stack install --install-ghc --local-bin-path bin
Downloading lts-18.24 build plan ...
RedownloadInvalidResponse Request {
  host                 = "raw.githubusercontent.com"
  port                 = 443
  secure               = True
  requestHeaders       = []
  path                 = "/fpco/lts-haskell/master//lts-18.24.yaml"
  queryString          = ""
  method               = "GET"
  proxy                = Nothing
  rawBody              = False
  redirectCount        = 10
  responseTimeout      = ResponseTimeoutDefault
  requestVersion       = HTTP/1.1
}
 "/root/.stack/build-plan/lts-18.24.yaml" (Response {responseStatus = Status {statusCode = 404, statusMessage = "Not Found"}, responseVersion = HTTP/1.1, responseHeaders = [("Connection","keep-alive"),("Content-Length","14"),("Content-Security-Policy","default-src 'none'; style-src 'unsafe-inline'; sandbox"),("Strict-Transport-Security","max-age=31536000"),("X-Content-Type-Options","nosniff"),("X-Frame-Options","deny"),("X-XSS-Protection","1; mode=block"),("Content-Type","text/plain; charset=utf-8"),("X-GitHub-Request-Id","DBD2:0C15:16CF9A:222CBC:62C7DB03"),("Accept-Ranges","bytes"),("Date","Fri, 08 Jul 2022 07:21:39 GMT"),("Via","1.1 varnish"),("X-Served-By","cache-nrt-rjtf7700044-NRT"),("X-Cache","MISS"),("X-Cache-Hits","0"),("X-Timer","S1657264899.445001,VS0,VE328"),("Vary","Authorization,Accept-Encoding,Origin"),("Access-Control-Allow-Origin","*"),("X-Fastly-Request-ID","0b37bcc0ad2d98420f6aaf39cc16e3d4524190eb"),("Expires","Fri, 08 Jul 2022 07:26:39 GMT"),("Source-Age","0")], responseBody = (), responseCookieJar = CJ {expose = []}, responseClose' = ResponseClose})
make: *** [Makefile:7: sv2v] Error 1
```

hello-eternity, 2022-07-08:
Is that because my internet connect error?

alibajelan, 2022-07-08:
Just upgrade stack
$stack upgrade --force-download

> Here is the error log, How can I sovle this
>
> ```
> mkdir -p bin
> stack install --install-ghc --local-bin-path bin
> Downloading lts-18.24 build plan ...
> RedownloadInvalidResponse Request { [...]

Upgrade stack
$stack upgrade --force-download

hello-eternity, 2022-07-09:
You are right. My stack version is too low. Thank you very much.

closed 2022-07-09 by hello-eternity