zachjs/sv2v #220: sv2v hangs and eventually OOM

flaviens, 2022-10-17:
Hi there!

I encountered a sv2v hang that eventually goes OOM (requires 100+GB RAM).
I could narrow it down to the following commands:

```
git clone https://github.com/black-parrot/black-parrot.git
cd black-parrot
git submodule update --init --recursive
sv2v -I ./external/HardFloat/source/RISCV/ -I./external/HardFloat/source/ -I./external/basejump_stl/bsg_misc/ -I./external/basejump_stl/bsg_noc/ -Ibp_common/src/include -Ibp_fe/src/include bp_fe/src/include/bp_fe_pkg.sv bp_common/src/include/bp_common_pkg.sv bp_be/src/include/bp_be_pkg.sv bp_fe/src/v/bp_fe_top.sv
```

It seems that this is not the only problematic occurrence in this repository, but the problem is probably common to all of these occurrences.

Thank you!
Flavien

zachjs, 2022-10-19:
Thanks for reporting this issue and providing a reproducer!

I have isolated the problem to the `always_comb` changes made in #199. I have not yet been able to track down the root cause or implement a fix. In the meantime, you may be able to workaround the issue using `-E always` if your downstream target supports the SystemVerilog `always` variants.

flaviens, 2022-10-20:
Hi Zach,

It indeed solved the problem for the narrowed down instance, thank you!

However, with the attached more real-world example [bp_pickled.zip](https://github.com/zachjs/sv2v/files/9831351/bp_pickled.zip), it still seems to still take at least a lot of memory:

```
sv2v -DSYNTHESIS -DVERILATOR -E always bp_pickled.sv -w sv2v_out.v
```

zachjs, 2022-12-24:
I've pushed some fixes and optimizations. With those changes, the `-E always` workaround isn't needed, and your most recent example converts in under 90 seconds on my machine. Can you give this a shot and let me know if it works for you?

flaviens, 2023-01-05:
Cool, thanks! In fact I'll continue using -E always because usually downstream tools I'm seem to work fine with that at the moment.
That said, this is a great addition for Verilog compliance.

closed 2023-01-05 by flaviens