zachjs/sv2v #222: Unknown bindings  port connections

hello-eternity, 2022-10-24:
Sorry, I can't find the bug in detail, and here is my error log 
Do I need to change some syntax？
```
sv2v: unknown bindings "rt_cid_delta_seed_i", "rt_cid_base_seed_i" specified for port connections in instance "rubytest_top_u" of "rubytest_top"
CallStack (from HasCallStack):
  error, called at src/Convert/ResolveBindings.hs:111:9 in main:Convert.ResolveBindings
```
and code the top is top.sv 

[code.zip](https://github.com/zachjs/sv2v/files/9851089/code.zip)

zachjs, 2022-10-29:
sv2v preprocesses files in the order in which they are specified, in line with other tools I have encountered. In this case, the macro definitions in `rubytest_define.sv` need to be processed before any files which depend on them. Thus `sv2v *.sv` produces the error above, but `sv2v rubytest_core.sv *.sv` seems to work as expected. Please let me know if this works for you!

The portions of the relevant files are processed in the order below when using `sv2v *.sv`, leading to the unknown port connection which sv2v was complaining about.

`rubytest_core.sv`:
```SystemVerilog
...
`ifdef RT_MODE_CLASSIC
,input logic [RT_CID_DELTA_NUM_W-1:0] rt_cid_delta_seed_i//2
,input logic [RT_CHECK_NUM_W-1:0] rt_cid_base_seed_i//5
`else
,input logic [RT_CHECK_GEN_ADDR_W-1:0] rt_info_addr_seed_i
,input logic [$bits(lsu_op_e)-1:0]     rt_info_opcode_seed_i
`endif
...
```

`rubytest_define.sv`:
```SystemVerilog
...
`define RT_MODE_CLASSIC
...
```

`top.sv`:
```SystemVerilog
...
  `ifdef RT_MODE_CLASSIC
    .rt_cid_delta_seed_i        (_rt_cid_delta_seed),
    .rt_cid_base_seed_i         (_rt_cid_base_seed),
  `else
    .rt_info_addr_seed_i        (_rt_info_addr_seed),
    .rt_info_opcode_seed_i      (_rt_info_opcode_seed),
  `endif
...
```

hello-eternity, 2022-11-01:
That worked！ Thank you for your patience

closed 2022-11-01 by hello-eternity