zachjs/sv2v #226: Issue when converting CV32E40X

mole99, 2023-01-31:
Hello,

I am trying to convert the [cv32e40x](https://github.com/openhwgroup/cv32e40x) core to Verilog with the following command: `sv2v -v rtl/include/cv32e40x_pkg.sv rtl/*.sv > processed.v`

The command completes without error, but when I take a look at the output file I see that the toplevel module has been removed (together with some other modules):

```
// removed module with interface ports: cv32e40x_core
```

These modules make extensive use of modports.

Is there any way to solve this issue?

Thanks!

zachjs, 2023-01-31:
Because `cv32e40x_core` has ports which are modports, sv2v must be able to observe and convert its instantiation. Modules with modport or interface ports which are uninstantiated cannot necessarily be represented with equivalent Verilog, so sv2v requires that the module and its instantiation be converted at the same time. Please try adding a wrapper or top-level module which instantiates the core.

mole99, 2023-02-01:
Thanks for your reply!

I added a wrapper `cv32e40x_top` around `cv32e40x_core` where the interface `if_xif` is instantiated and signals are assigned. The modports are then passed to the `cv32e40x_core`.

The conversion completes without error, and it seems to be a success!

Still the modules are removed:

```
// removed module with interface ports: cv32e40x_core
```

But the logic of `cv32e40x_core`  is now contained inside `cv32e40x_top`.

The only catch: The tool did not remove/resolve some of the typedefs in `cv32e40x_top`, for example `x_compressed_req_t` on line 9330 in `preprocessed.v`.

Presumably because they were declared inside the interface `if_xif`.

I pushed my code to [mole99/cv32e40x](https://github.com/mole99/cv32e40x) if you want to take a quick look at it.
You can run the conversion with `make preprocessed.v`.

After adding these typedefs manually I could run synthesis on yosys. Your tool is really awesome!

zachjs, 2023-02-12:
Thanks for sharing! I'm glad the flow seems to be working.

Within `cv32e40x_top`, `x_compressed_req_t` and the other unconverted typenames are defined within `if_xif`, and so are not in scope. If you need to use those types, I think the traditional solution would be to put the typedefs in a package that both the interface and the module can import.

SystemVerilog also has a concept called "interface based typedefs" which can be used (or abused) as a workaround. IEEE 1800-2017 Section 6.18 has the following simple example.
```SystemVerilog
interface intf_i;
    typedef int data_t;
endinterface
module sub(intf_i p);
    typedef p.data_t my_data_t;
    my_data_t data;
    // type of 'data' will be int when connected to interface above
endmodule
```

In your specific case, you might do something like the below, where the interface based typedef refers to a type within an *interface instance* rather than a *bound modport*. **Please note:** the standard doesn't necessarily say that this should work. sv2v and at least one commercial toolchain support it, but many tools do not.
```SystemVerilog
    if_xif my_if_xif();
    typedef my_if_xif.x_compressed_req_t x_compressed_req_t;
    x_compressed_req_t compressed_req;
```

mole99, 2023-02-13:
I have added all typedefs according to the interface.

The design is now converted from SystemVerilog to Verilog without manual intervention. This is exactly what I was looking for!

From my side this issue is resolved and can be closed. Thanks a lot @zachjs!

zachjs, 2023-02-21:
I'm glad to hear that's it working for you! Please feel free to file a new issue or reopen this one if you run into problems.

closed 2023-02-21 by zachjs

MBleaux, 2024-07-05:
Hello @mole99 would you mind sharing the cv32e40x_top file that you used to solve the conversion interface issue? Because I'm facing the same problem with the interfaces when I try to convert the cv32e40s. I read this conversation but I couldn't fix my problem, neither found cv32e40x_top file on your repository as you mentioned to compare :(

mole99, 2024-07-05:
Hi @MBleaux, you should be able to find it here: https://github.com/semify-eda/tristan/blob/2cc920f8c6ea28f4ec096ee60b92592b67300be9/core/cv32e40x_top.sv

Just in case the link breaks in the future, here's some code:

<details>
  <summary>cv32e40x_top.sv</summary>

```SystemVerilog
`default_nettype none

module cv32e40x_top import cv32e40x_pkg::*;
#(
    parameter INSTR_RDATA_WIDTH = 32,
              BOOT_ADDR         = 32'h02000000 + 24'h200000,
              DM_HALTADDRESS    = 32'h1A11_0800,
              HART_ID           = 32'h0000_0000,
              NUM_MHPMCOUNTERS  = 1
)
(
    // Clock and reset
    input  logic                          clk_i,
    input  logic                          rst_ni,

    // Instruction memory interface
    output logic                          instr_req_o,
    input  logic                          instr_gnt_i,
    input  logic                          instr_rvalid_i,
    output logic [31:0]                   instr_addr_o,
    input  logic [31:0]                   instr_rdata_i,

    // Data memory interface
    output logic                          data_req_o,
    input  logic                          data_gnt_i,
    input  logic                          data_rvalid_i,
    output logic [31:0]                   data_addr_o,
    output logic [3:0]                    data_be_o,
    output logic                          data_we_o,
    output logic [31:0]                   data_wdata_o,
    input  logic [31:0]                   data_rdata_i,

    // Cycle count
    output logic [63:0]                   mcycle_o,

    // eXtension interface
    //if_xif.cpu_compressed                 xif_compressed_if,
    //if_xif.cpu_issue                      xif_issue_if,
    //if_xif.cpu_commit                     xif_commit_if,
    //if_xif.cpu_mem                        xif_mem_if,
    //if_xif.cpu_mem_result                 xif_mem_result_if,
    //if_xif.cpu_result                     xif_result_if,

    // Debug interface
    input  logic                          debug_req_i,

    // CPU control signals
    input  logic                          fetch_enable_i,
    output logic                          core_sleep_o
);

    localparam X_NUM_RS = 2;

    cv32e40x_if_xif #(
        .X_NUM_RS    ( X_NUM_RS ),
        .X_MEM_WIDTH ( 32 ),
        .X_RFR_WIDTH ( 32 ),
        .X_RFW_WIDTH ( 32 ),
        .X_MISA      ( '0 )
    ) ext_if();

    coproc coproc_inst
[... 192 lines cut ...]
                id  <= xif_issue_if.issue_req.id;
                cntb_start <= 1'b1;
            end
        end
    end

    logic [4:0] cntb_result;

    cntb cntb_inst
    (
        .clk_i      (clk_i      ),
        .rst_ni     (rst_ni     ),
        .start_i    (cntb_start ),
        .word_i     (rs0        ),
        .index_i    (rs1[4:0]   ),
        .result_o   (cntb_result),
        .done_o     (cntb_done  )
    );
  
endmodule
```

</details>

Let me know if there are any issues!

MBleaux, 2024-07-10:
It worked fine, thank you for sharing the files and links!

mole99, 2024-07-10:
That's great to hear!