zachjs/sv2v #227: Why "sv2v: src\Convert\TypeOf.hs:417:11-37: Non-exhaustive patterns in (tf, _ : rs)" ?

forthyen, 2023-02-08:
sv2v -v cv32e40xx.sv

and then

sv2v: src\Convert\TypeOf.hs:417:11-37: Non-exhaustive patterns in (tf, _ : rs)

[cv32e40xx.zip](https://github.com/zachjs/sv2v/files/10682552/cv32e40xx.zip)

forthyen, 2023-02-08:
Sorry!  I add -DMAIN_CORE_CV32E40X and modify source.  Upload it again !

[cv32e40xx.zip](https://github.com/zachjs/sv2v/files/10682650/cv32e40xx.zip)

closed 2023-02-08 by forthyen

zachjs, 2023-02-13:
@forthyen Is there an issue you're still facing here? Even if a failure like this is caused by an error in the source (it might be not be!), I think sv2v should be able to do a better job communicating that issue to the user.

I can reproduce the failure locally. If you have a workaround, it would be great to know more, especially if others run into a similar issue.