zachjs/sv2v #229: Convertion unsized single-bit value

kele14x, 2023-02-21:
IEEE 2800-2017 5.7.1 Specifies unsized single-bit value as "set all bit". But sv2v write out signle bit value (`'0 -> 1'sb0` and `'1 -> 1'sb1`).

> An unsized single-bit value can be specified by preceding the single-bit value with an apostrophe ( ' ), but without the base specifier. All bits of the unsized value shall be set to the value of the specified bit. In a self-determined context, an unsized single-bit value shall have a width of 1 bit, and the value shall be treated as unsigned.
  '0, '1, 'X, 'x, 'Z, 'z // sets all bits to specified value

Tested with sv2v v0.0.10-0-g87642c0 on Windows.

zachjs, 2023-02-21:
The resulting expression for these literals depends on the context in which they are used. It sounds like you may have found a case where the converted result is not behaviorally equivalent to the source. Could you share a concrete test case so that I can work on this issue? Thank you!

kele14x, 2023-02-21:
Hi,

Seems GitHub does no allow me directly upload .sv file. So I compress file **issue_test.sv** into a zip:

[issue_test.zip](https://github.com/zachjs/sv2v/files/10795948/issue_test.zip)

The issue was on line 11

``` systemverilog
logic [7:0] dout;
...
dout <= '1;
```

`dout` is a 8-bit logic value, which will be reset to `8'hFF` when `rst` is assert. But generated **.v** file set it to 1:

```Verilog
dout <= 1'sb1;
```

zachjs, 2023-02-22:
In this particular context, `1'sb1` is functionally equivalent to `'1` (and `8'hFF`) due to sign-extension. sv2v uses `1'sb1` as an optimization, as it is equivalent to `'1` regardless of the type of the left-hand side.

As a basic example, the below prints `111111` as desired.
```SystemVerilog
module top;
  logic [5:0] x;
  initial begin
    x = 1'sb1;
    $display("%b", x);
  end
endmodule
```

kele14x, 2023-02-22:
Thanks @zachjs, I made a mistake on purpose of `'sd`. It should work as desired.

closed 2023-02-22 by kele14x