zachjs/sv2v #231: Issue in how block comments are handled

mole99, 2023-03-06:
There seems to be a small bug in how `sv2v` handles block comments :)

Take for example this comment:

```
/*/This is a test*/
```

`sv2v` generates this error: `test.sv:1:18: Parse error: unexpected token '*' (Sym_aster)`

While all of these comments are fine:

```
/*This is a test*/
```

```
/*This is a test/*/
```

```
/* /This is a test/*/
```

According to the small section about comments in the SV LRM everything between `/*` and `*/` should be ignored:

[image]

zachjs, 2023-03-08:
Thanks for reporting this issue! I agree with your interpretation of the standard. I just pushed a change that should fix this behavior. Please let me know if it works for you!

mole99, 2023-03-09:
I can confirm that it works!

Thanks for the quick solution :)

closed 2023-03-09 by mole99