zachjs/sv2v #233: interface to verilog, package files, typedef and enums

armstrong0, 2023-04-13:
I have a verilog project:   https://github.com/armstrong0/PDP8e  it uses yosys to target lattice ICE40 fpga.  I am adding a subsystem to it that emulates a rk8e with rk05 drives using a sdcard as storage.  Rather than write my own I have borrowed a design from a project on opencores by Rob Doyle.  His code is in VHDL, I have manually translated to systemverilog.  I intend to use sv2v to convert it to verilog so it can be included in the FPGA build.  I have pasted one of the header files below.
In particular when sv2v is run all information about the enums disappears.  It would be useful to be able to process the header files into verilog where the enum values become  named pairs:

parameter sdopNOP = 0;

Then the higher level file can include the file and be able to use sensible names in the interface.

sv2v will not currently process package files.

package sd_types;

  // Types
  typedef logic [0:7] sdBYTE_t;  // Byte
  typedef sdBYTE_t [0:5] sdCMD_t;  // SD Commands
  typedef logic sdLEN_t;  // Read/Write Length
  typedef logic [0:14] addr_t;  // 12 bit memory address type
  // added because I did not want to import too many things
  typedef logic [0:31] sdDISKaddr_t;  // SD Sector Address
  typedef logic [0:6] sdCCRC_t;  // Command CRC
  typedef logic [0:15] sdDCRC_t;  // Data CRC

  typedef enum logic [1:0] {
    sdopNOP,    // SD NOP
    sdopABORT,  // Abort Read or Write
    sdopRD,     // Read SD disk
    sdopWR      // write to disk
  } sdOP_t;
  typedef enum logic [2:0] {
    sdstateINIT,    // SD Initializing
    sdstateREADY,   // SD Ready for commands
    sdstateREAD,    // SD Reading
    sdstateWRITE,   // SD Writing
    sdstateDONE,    // SD Done
    sdstateINFAIL,  // SD Initialization Failed
    sdstateRWFAIL
  }  // SD Read/Write Failed
  sdSTATE_t;
  typedef struct packed {
    sdSTATE_t state;  // SD Status
    sdBYTE_t  err;    // Error Status
    sdBYTE_t  val;    // Value Status
    sdBYTE_t  rdCNT;  // Read Count Status
    sdBYTE_t  wrCNT;  // Write Count Status
    sdBYTE_t  debug;
  } sdSTAT_t;  // Debug State

zachjs, 2023-04-13:
Can you try passing all of your SystemVerilog source files into sv2v all at once? sv2v relies on, for example, being able to see the definition of a package and the modules that use that package simultaneously.

zachjs, 2023-04-23:
@armstrong0 Did you have any luck with the above suggestion? Either way, I'm eager for your feedback!

armstrong0, 2023-04-23:
Yes, I have changed my wrapper unit from verilog to systemverilog.  So now enums and structs don't need to be referenced from verilog.  I have 3 sv files and two headers and a verilog test bench all happily working together in simulation with iverilog.  I have not got to the point of integrating it with the rest of the verilog code but I am pretty confident it will work.

Dave

zachjs, 2023-05-18:
@armstrong0 Were you able to get this working?

zachjs, 2023-05-31:
I'm closing because I think the original issue is now resolved. Please feel free to reopen or file a new issue if you have further questions!

I have since added the following language to the readme to clarify this aspect of sv2v's usage.

> Users should typically pass all of their SystemVerilog source files to sv2v at once so it can properly resolve packages, interfaces, type parameters, etc., across files.

closed 2023-05-31 by zachjs