zachjs/sv2v #240: Problems installing

philn128, 2023-05-26:
I don't know any Haskell so this may be a simple fix. I'm on Ubuntu 20.04.3

I'm on `master` commit `0a7b0250e7392dab884f70fad75055782a781da8`

I'm running the following:

```bash
sudo apt install haskell-stack
git clone https://github.com/zachjs/sv2v.git
cd sv2v
make
```

I get the following error:

```bash
mkdir -p bin
stack install --install-ghc --local-bin-path bin
Downloading lts-18.24 build plan ...
RedownloadInvalidResponse Request {
  host                 = "raw.githubusercontent.com"
  port                 = 443
  secure               = True
  requestHeaders       = []
  path                 = "/fpco/lts-haskell/master//lts-18.24.yaml"
  queryString          = ""
  method               = "GET"
  proxy                = Nothing
  rawBody              = False
  redirectCount        = 10
  responseTimeout      = ResponseTimeoutDefault
  requestVersion       = HTTP/1.1
}
 "/home/lab/.stack/build-plan/lts-18.24.yaml" (Response {responseStatus = Status {statusCode = 404, statusMessage = "Not Found"}, responseVersion = HTTP/1.1, responseHeaders = [("Connection","keep-alive"),("Content-Length","14"),("Content-Security-Policy","default-src 'none'; style-src 'unsafe-inline'; sandbox"),("Strict-Transport-Security","max-age=31536000"),("X-Content-Type-Options","nosniff"),("X-Frame-Options","deny"),("X-XSS-Protection","1; mode=block"),("Content-Type","text/plain; charset=utf-8"),("X-GitHub-Request-Id","FE9E:3F6D:680B0D:797652:64701DD5"),("Accept-Ranges","bytes"),("Date","Fri, 26 May 2023 02:47:49 GMT"),("Via","1.1 varnish"),("X-Served-By","cache-den8277-DEN"),("X-Cache","MISS"),("X-Cache-Hits","0"),("X-Timer","S1685069269.430220,VS0,VE107"),("Vary","Authorization,Accept-Encoding,Origin"),("Access-Control-Allow-Origin","*"),("X-Fastly-Request-ID","79e67ac6591147b941b1ef086cf400ca75f7846c"),("Expires","Fri, 26 May 2023 02:52:49 GMT"),("Source-Age","0")], responseBody = (), responseCookieJar = CJ {expose = []}, responseClose' = ResponseClose})
make: *** [Makefile:7: sv2v] Error 1
```

Any ideas?

philn128, 2023-05-26:
title

zachjs, 2023-05-26:
What version of stack do you have installed? Please try `stack upgrade`.

philn128, 2023-05-26:
I was using `1.9.3.1 x86_64`. After running `sudo stack upgrade` the version number is now `2.11.1` and the `make` command seems to work since I can run `bin/sv2v --help`.

I'm looking forward to trying it out!

closed 2023-05-26 by philn128