zachjs/sv2v #242: make test fails

dwRchyngqxs, 2023-05-29:
`make test` fails:
There are warnings as described in #241 and some tests fails ([the standard output log](https://github.com/zachjs/sv2v/files/11590247/report.txt)).

zachjs, 2023-05-29:
I pushed e1948689dd6d27a2bf396e48ddc88e2db8fb170e to a temporary branch `upgrade-iverilog`. It should address the incompatibilities with newer versions of iverilog. At least one test still fails with the latest version of iverilog due to https://github.com/steveicarus/iverilog/issues/927. Does this work for you?

zachjs, 2023-05-30:
I merged these changes into master and removed the temporary branch. I will update the title to reflect the remaining work: I should bump the iverilog version further once crash is fixed.

zachjs, 2023-06-11:
The iverilog issue is now fixed. I have bumped the iverilog version and confirmed all tests pass. Thank you for pointing this issue out to me!

closed 2023-06-11 by zachjs