zachjs/sv2v #245: error:      warning: passing 'char *' to parameter of type 'const unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]

yurivict, 2023-06-05:
The Haskell compiler prints errors or warnings:

```
cd /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v &&  /bin/ln -fs /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/_cabal_deps &&  /bin/ln -fs ../cabal.project.local &&  /usr/bin/env DESTDIR=  HOME=/usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01 XDG_DATA_HOME=/usr/ports/cad/yosys-systemverilog/work  XDG_CONFIG_HOME=/usr/ports/cad/yosys-systemverilog/work  XDG_CACHE_HOME=/usr/ports/cad/yosys-systemverilog/work/.cache  HOME=/usr/ports/cad/yosys-systemverilog/work PATH=/usr/local/libexec/ccache:/usr/ports/cad/yosys-systemverilog/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin PKG_CONFIG_LIBDIR=/usr/ports/cad/yosys-systemverilog/work/.pkgconfig:/usr/local/libdata/pkgconfig:/usr/local/share/pkgconfig:/usr/libdata/pkgconfig MK_DEBUG_FILES=no MK_KERNEL_SYMBOLS=no SHELL=/bin/sh NO_LINT=YES PREFIX=/usr/local  LOCALBASE=/usr/local  CC="cc" CFLAGS="-O2 -pipe  -fstack-protector-strong -fno-strict-aliasing "  CPP="cpp" CPPFLAGS="-I/usr/local/include"  LDFLAGS=" -L/usr/local/lib -fstack-protector-strong " LIBS=""  CXX="c++" CXXFLAGS="-O2 -pipe -fstack-protector-strong -fno-strict-aliasing  "  MANPREFIX="/usr/local" CCACHE_DIR="/tmp/.ccache" BSD_INSTALL_PROGRAM="install  -s -m 555"  BSD_INSTALL_LIB="install  -s -m 0644"  BSD_INSTALL_SCRIPT="install  -m 555"  BSD_INSTALL_DATA="install  -m 0644"  BSD_INSTALL_MAN="install  -m 444" XDG_DATA_HOME=/usr/ports/cad/yosys-systemverilog/work/cabal-home XDG_CONFIG_HOME=/usr/ports/cad/yosys-systemverilog/work/cabal-home XDG_CACHE_HOME=/usr/ports/cad/yosys-systemverilog/work/cabal-home HOME=/usr/ports/cad/yosys-systemverilog/work/cabal-home cabal build --offline --disable-benchmarks --disable-tests   --flags ""  exe:sv2v
Warning: No remote package servers have been specified. Usually you would have
one specified in the config file.
Resolving dependencies...
Build profile: -w ghc-9.4.5 -O1
In order, the following will be built (use -v for more details):
 - alex-3.3.0.0 (exe:alex) (first run)
 - cmdargs-0.10.22 (lib) (first run)
 - happy-1.20.1.1 (exe:happy) (first run)
 - hashable-1.4.2.0 (lib) (first run)
 - primitive-0.8.0.0 (lib) (first run)
 - th-compat-0.1.4 (lib) (first run)
 - vector-stream-0.1.0.0 (lib) (first run)
 - githash-0.1.6.3 (lib) (first run)
 - vector-0.13.0.0 (lib) (first run)
 - sv2v-0.0.10 (exe:sv2v) (first run)
Configuring executable 'alex' for alex-3.3.0.0..
Configuring executable 'happy' for happy-1.20.1.1..
Configuring library for hashable-1.4.2.0..
Configuring library for vector-stream-0.1.0.0..
Configuring library for primitive-0.8.0.0..
Configuring library for th-compat-0.1.4..
Configuring library for cmdargs-0.10.22..
Preprocessing library for cmdargs-0.10.22..
Preprocessing executable 'happy' for happy-1.20.1.1..
Preprocessing library for hashable-1.4.2.0..
Building library for hashable-1.4.2.0..
Building library for cmdargs-0.10.22..
Building executable 'happy' for happy-1.20.1.1..
Preprocessing library for vector-stream-0.1.0.0..
Preprocessing executable 'alex' for alex-3.3.0.0..
Building library for vector-stream-0.1.0.0..
Preprocessing library for th-compat-0.1.4..
Preprocessing library for primitive-0.8.0.0..
Building library for th-compat-0.1.4..
Building library for primitive-0.8.0.0..
Building executable 'alex' for alex-3.3.0.0..
[[1 of 1] Compiling Data.Stream.Monadic ( src/Data/Stream/Monadic.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/vector-stream-0.1.0.0/build/Data/Stream/Monadic.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/vector-stream-0.1.0.0/build/Data/Stream/Monadic.dyn_o )
1 of 1] Compiling Language.Haskell.TH.Syntax.Compat ( src/Language/Haskell/TH/Syntax/Compat.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/th-compat-0.1.4/build/Language/Haskell/TH/Syntax/Compat.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/th-compat-0.1.4/build/Language/Haskell/TH/Syntax/Compat.dyn_o )
[ 1 of 25] Compiling Data.Generics.Any ( Data/Generics/Any.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/cmdargs-0.10.22/build/Data/Generics/Any.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/cmdargs-0.10.22/build/Data/Generics/Any.dyn_o )
[1 of 7] Compiling Data.Hashable.Imports ( src/Data/Hashable/Imports.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/hashable-1.4.2.0/build/Data/Hashable/Imports.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/hashable-1.4.2.0/build/Data/Hashable/Imports.dyn_o )
[2 of 7] Compiling Data.Hashable.LowLevel ( src/Data/Hashable/LowLevel.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/hashable-1.4.2.0/build/Data/Hashable/LowLevel.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/hashable-1.4.2.0/build/Data/Hashable/LowLevel.dyn_o )
[ 1 of 13] Compiling Control.Monad.Primitive ( Control/Monad/Primitive.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/primitive-0.8.0.0/build/Control/Monad/Primitive.o, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/primitive-0.8.0.0/build/Control/Monad/Primitive.dyn_o )
[ 1 of 19] Compiling AbsSyn           ( src/AbsSyn.lhs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/happy-1.20.1.1/x/happy/build/happy/happy-tmp/AbsSyn.o )

/tmp/ghc93701_0/ghc_10.c:10:164: error:
     warning: passing 'char *' to parameter of type 'const unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
   |
10 | HsWord64 ghczuwrapperZC1ZChashablezm1zi4zi2zi0zminplaceZCDataziHashableziLowLevelZChashablezufnvzuhash(char* a1, HsInt64 a2, HsInt64 a3) {return hashable_fnv_hash(a1, a2, a3);}
   |                                                                                                                                                                    ^
HsWord64 ghczuwrapperZC1ZChashablezm1zi4zi2zi0zminplaceZCDataziHashableziLowLevelZChashablezufnvzuhash(char* a1, HsInt64 a2, HsInt64 a3) {return hashable_fnv_hash(a1, a2, a3);}
                                                                                                                                                                   ^~

include/HsHashable.h:19:53: error:
     note: passing argument to parameter 'str' here
   |
19 | FNV_UNSIGNED hashable_fnv_hash(const unsigned char* str, FNV_SIGNED len, FNV_UNSIGNED salt);
   |                                                     ^
FNV_UNSIGNED hashable_fnv_hash(const unsigned char* str, FNV_SIGNED len, FNV_UNSIGNED salt);
                                                    ^
1 warning generated.
[ 1 of 22] Compiling Data.Ranged.Boundaries ( src/Data/Ranged/Boundaries.hs, /usr/ports/cad/yosys-systemverilog/work/yosys-systemverilog-cfd5801-2023-06-01/sv2v/dist-newstyle/build/x86_64-freebsd/ghc-9.4.5/alex-3.3.0.0/x/alex/build/alex/alex-tmp/Data/Ranged/Boundaries.o )

/tmp/ghc93701_0/ghc_17.c:10:164: error:
     warning: passing 'char *' to parameter of type 'const unsigned char *' converts between pointers to integer types where one is of the unique plain 'char' type and the other is not [-Wpointer-sign]
   |
10 | HsWord64 ghczuwrapperZC1ZChashablezm1zi4zi2zi0zminplaceZCDataziHashableziLowLevelZChashablezufnvzuhash(char* a1, HsInt64 a2, HsInt64 a3) {return hashable_fnv_hash(a1, a2, a3);}
[... cut ...]
                                                                                                                                                                   ^~

include/HsHashable.h:19:53: error:
     note: passing argument to parameter 'str' here
   |
19 | FNV_UNSIGNED hashable_fnv_hash(const unsigned char* str, FNV_SIGNED len, FNV_UNSIGNED salt);
   |                                                     ^
FNV_UNSIGNED hashable_fnv_hash(const unsigned char* str, FNV_SIGNED len, FNV_UNSIGNED salt);
                                                    ^
1 warning generated.
```

Version: 2023-06-01
ghc-9.4.5
FreeBSD 13.2

zachjs, 2023-06-06:
That output suggests you may be building sv2v in a non-standard way. Those errors are in compiling sv2v's dependencies, rather than sv2v itself. The "officially supported" method is to install Haskell Stack (https://docs.haskellstack.org/en/stable/), and then just run `make`. Can you give that a shot?

yurivict, 2023-06-06:
> Can you give that a shot?

I need to build it as part of the FreeBSD port build process, which means no downloads are allowed. Unfortunately, Stack downloads things during build. So I use a standard Haskell-based pipeline, which uses cabal and all the same dependencies that Stack would download.

zachjs, 2023-06-06:
I think the GHC and dependency versions referenced in the error message don't match the versions that stack is using.
* The error mentions `ghc-9.4.5`, but I'm using GHC `8.10.7`.
* The error mentions `alex-3.3.0.0`, but I'm using `alex-3.2.7.1`.
* etc.

You could try using the Hackage tarball from https://hackage.haskell.org/package/sv2v-0.0.10 rather than the latest git HEAD from GitHub. That tarball contains an sv2v.cabal with more autogenerated more specific version requirements. During the release process, I generate that tarball using `stack sdist`.

I'm also open to patches if, e.g., the upper bound on the version of `alex` is too loose. I'm also not opposed to upgrading to GHC 9.2.7 using the latest LTS Stackage.

yurivict, 2023-06-06:
ghc-9.4.5 builds sv2v fine. I reported the error/warning because it looks weird, but it isn't fatal.

I am able to create the port the way how it is now.

Please feel free to close it.

closed 2023-06-10 by zachjs