zachjs/sv2v #246: Struct parameters in module definition

lpawelcz, 2023-06-21:
Hi @zachjs, I tried to use sv2v to convert [VeeR EL2](https://github.com/chipsalliance/Cores-VeeR-EL2) core to Verilog but I came across this behaviour where sv2v does not convert correctly module parameters which are included from header files and are defined as struct.

You can reproduce the issue with the following example.

directory structure:
```
.
├── el2_lib.sv
└── inc
    ├── el2_param.vh
    └── el2_pdef.vh
```

inc/el2_pdef.vh:
```systemverilog
typedef struct packed {
	bit [8:0]    BTB_ADDR_HI;
	bit [8:0]    BTB_BTAG_SIZE;
} el2_param_t;
```

inc/el2_param.vh:
```systemverilog
parameter el2_param_t pt = '{
	BTB_ADDR_HI      : 9'h005,
	BTB_BTAG_SIZE    : 9'h009
}
```

el2_lib.sv:
```systemverilog
module el2_btb_tag_hash #(
    `include "el2_param.vh"
) (
    input logic [pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+1] pc,
    output logic [pt.BTB_BTAG_SIZE-1:0] hash
);

    assign hash = {(
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+1] ^
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+1] ^
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+1]
	)};
endmodule
```

Here is the result of converting `el2_lib.sv` with `sv2v -I ./inc -w adjacent el2_lib.sv` to `el2_lib.v`:
```verilog
module el2_btb_tag_hash (
	pc,
	hash
);
	parameter el2_param_t pt = '{
		BTB_ADDR_HI: 9'h005,
		BTB_BTAG_SIZE: 9'h009
	};
	input wire [((pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI + 1] pc;
	output wire [pt.BTB_BTAG_SIZE - 1:0] hash;
	assign hash = {(pc[((pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE:((pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE) + 1] ^ pc[(pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE) + pt.BTB_BTAG_SIZE:(pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE) + 1]) ^ pc[pt.BTB_ADDR_HI + pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI + 1]};
endmodule
```

For more context here is the syntax error from yosys:
```
el2_lib.v:5: ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ';'
```
I believe parameter `pt` should be unfolded into separate parameters: `BTB_ADDR_HI` and `BTB_BTAG_SIZE` and the references in the module should also take this into account.

Another thing worth noting about this issue is that it is not strictly related to parameter being included inside the module definition because we can modify `el2_lib.sv` and manually include the param like so:
```systemverilog
module el2_btb_tag_hash #(
    parameter el2_param_t pt = '{
        BTB_ADDR_HI      : 9'h005,
        BTB_BTAG_SIZE    : 9'h009
    }
) (
    input logic [pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+1] pc,
    output logic [pt.BTB_BTAG_SIZE-1:0] hash
);

    assign hash = {(
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE+1] ^
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE+1] ^
        pc[pt.BTB_ADDR_HI+pt.BTB_BTAG_SIZE:pt.BTB_ADDR_HI+1]
	)};
endmodule
```

and we would still get exactly the same result in converted verilog.

zachjs, 2023-06-21:
Thank you for filing a detailed issue! In your example, it doesn't look like anything includes `el2_pdef.vh`. With the include added `el2_lib.sv`, I get:
```Verilog
module el2_btb_tag_hash (
	pc,
	hash
);
	parameter [17:0] pt = 18'h00a09;
	input wire [((pt[17-:9] + pt[8-:9]) + pt[8-:9]) + pt[8-:9]:pt[17-:9] + 1] pc;
	output wire [pt[8-:9] - 1:0] hash;
	assign hash = {(pc[((pt[17-:9] + pt[8-:9]) + pt[8-:9]) + pt[8-:9]:((pt[17-:9] + pt[8-:9]) + pt[8-:9]) + 1] ^ pc[(pt[17-:9] + pt[8-:9]) + pt[8-:9]:(pt[17-:9] + pt[8-:9]) + 1]) ^ pc[pt[17-:9] + pt[8-:9]:pt[17-:9] + 1]};
endmodule
```

lpawelcz, 2023-06-21:
That looks awesome, I'll surely test that out. Thanks a lot!

zachjs, 2023-07-02:
@lpawelcz Is there anything more to work on for this issue?

lpawelcz, 2023-07-04:
Oh sorry for the delay, I think we can close this issue as the problem was on my side. Thank you for your help, I really appreciate that. I also started using this new write directory feature and I must say it is super helpful :)

closed 2023-07-04 by lpawelcz