zachjs/sv2v #247: Issues with newest iverilog

dwRchyngqxs, 2023-07-06:
While running `make test` with latest iverilog, some test failed because of Verilog testbenches using SystemVerilog.
Should iverilog be called with the SystemVerilog flag or should the testbenches be fixed (I can submit a PR for that)?

Another issue was iverilog takes a very long time to compile the output of sv2v on `test/core/string_param.sv`. I don't know what's causing it but that's an iverilog issue, not an sv2v issue.

zachjs, 2023-07-08:
Thanks for highlighting this. I filed https://github.com/steveicarus/iverilog/issues/967 for the issue with `test/core/string_param.v`. I pushed d0e3b794bc11974a4261075acc82be9c10de27ef to bump iverilog as far as I could and fix the incompatibilities. The choice to run iverilog in Verilog-2005 mode is intentional, as sv2v should target this more restrictive standard.

zachjs, 2023-08-01:
The issue in iverilog has been resolved. I have bumped the CI version to the latest. Is there anything outstanding here?

dwRchyngqxs, 2023-08-01:
Everything is solved as far as I know.

closed 2023-08-01 by dwRchyngqxs