zachjs/sv2v #249: sv2v: unexpected non-var or non-port function decl

gadfort, 2023-07-26:
A user of https://github.com/siliconcompiler/siliconcompiler (https://github.com/siliconcompiler/siliconcompiler/issues/1789) reported that sv2v isn't able to parse one of their files, however, some other system verilog converters seem to be able to (but we don't support those).

I've attached their file: [encode_stream_sc_module_8.zip](https://github.com/zachjs/sv2v/files/12172787/encode_stream_sc_module_8.zip)

To run:
```
sv2v encode_stream_sc_module_8.sv
```
Log output for version: sv2v v0.0.10-29-g2bb9c82:
sv2v: unexpected non-var or non-port function decl: inout logic [2:0] ts_s__ref_0;
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux-tinfo6/Cabal-3.2.1.0/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:10710:3 in main:Language.SystemVerilog.Parser.Parse

If this is not something sv2v supports, please let me know.

zachjs, 2023-07-27:
Thank you for filing this issue! I just pushed a fix: bef9b1a3f06c9db3a1fef0e6d9489f7a4f8780f8. The given test case now parses cleanly.

gadfort, 2023-07-27:
@zachjs thank you

closed 2023-07-27 by gadfort