zachjs/sv2v #253: [feature request] support for field name preservation when struct conversion

Dragon-Git, 2023-08-28:
Sv2v version 0.11.0 is represented by vectors when converting the structure, and the name of the field cannot be seen in the converted Verilog code, resulting in poor readability, can the name information of the structure be retained in the converted Verilog code?
example：
```systemverilog
typedef struct {
  logic [31:0] a;
  logic [31:0] b;
  logic op;
} data_struct_t;

module src (
  output data_struct_t my_data 
  );
  assign my_data.a = 32'h12345678;
  assign my_data.b = 32'h87654321;
  assign my_data.op = 1'b1;
endmodule

module dest (
  input data_struct_t my_data,
  output logic result
  );
  assign result = my_data.op? my_data.a - my_data.b : my_data.b - my_data.a;
endmodule

module top;
  data_struct_t my_data;

  src src_inst (
    .my_data(my_data)
  );

  dest dest_inst (
    .my_data(my_data)
  );

endmodule

```
Now convert to:
```systemverilog
module src (my_data);
        output wire [64:0] my_data;
        assign my_data[64-:32] = 32'h12345678;
        assign my_data[32-:32] = 32'h87654321;
        assign my_data[0] = 1'b1;
endmodule
module dest (
        my_data,
        result
);
        input wire [64:0] my_data;
        output wire result;
        assign result = (my_data[0] ? my_data[64-:32] - my_data[32-:32] : my_data[32-:32] - my_data[64-:32]);
endmodule
module top;
        wire [64:0] my_data;
        src src_inst(.my_data(my_data));
        dest dest_inst(.my_data(my_data));
endmodule
```
Expected:
```systemverilog
module src (my_data);
        output wire [64:0] my_data;
        assign my_data_a = 32'h12345678;
        assign my_data_b = 32'h87654321;
        assign my_data_op = 1'b1;
endmodule
module dest (
        my_data,
        result
);
        input wire [64:0] my_data;
        output wire result;
        assign result = (my_data_op ? my_data_a - my_data_b : my_data_b - my_data_a);
endmodule
module top;
        wire [64:0] my_data;
        src src_inst(.my_data(my_data));
        dest dest_inst(.my_data(my_data));
endmodule
```

zachjs, 2023-08-29:
Thanks for filing this request! I agree that sv2v's struct conversion doesn't produce highly readable output. Many of sv2v's conversions exhibit this limitation to some extent. sv2v was envisioned as part of synthesis flows, producing generated output that didn't necessarily need to be readable. Can you help me understand your use case here? What do you intend to do with the converted output? Is there a pain point in your existing workflow, e.g., #194?

KatCe, 2023-09-18:
Hello. I could also need this feature for debugging purposes. We use sv2v in order to get a Verilog version of a design. Then we feed it into our Yosys-based flow, where we perform further analysis passes. Analysing the results is quite time consuming when the names are not preserved. Would appreciate it. Thank you!

sifferman, 2023-10-12:
I wanted to recommend that if this feature is added, I think it should be added only as an option. Arrays of structs are currently synthesized as a BRAM. But if an array of structs is separated into multiple different arrays, then extra BRAMs may be inferred.

Example:
```systemverilog
struct {
  logic a,
  logic b
} s[8];
```
```verilog
reg [1:0] s [0:7];
// or
reg s_a [0:7];
reg s_b [0:7];
```

KatCe, 2023-10-25:
Implementing it as an option sounds good. I also have another use case now, where I would need this feature for generating verification code.