zachjs/sv2v #255: name conflict

forthyen, 2023-09-24:
Would you please check it?

package pkg;
   function automatic int a;
	  return 1;
   endfunction : a
endpackage

typedef struct packed {
   logic a;
   logic b;
   logic c;
} type_t;

module b;
  import pkg::*;
  
  type_t ab = '{
   b:'1,
   a:'1,
   default: '0};
endmodule // b;

[test.zip](https://github.com/zachjs/sv2v/files/12707964/test.zip)

zachjs, 2023-09-24:
Can you be more specific about the behavior you're seeing compared to what you expect to see? If you can, this will help me address your issue more quickly.

zachjs, 2023-09-24:
I think I see the conflict. Can you point to source that exhibits this issue? Your test case is very helpful, but I want to avoid designing for just this specific case.

forthyen, 2023-09-24:
Thanks !

./sv2v test.sv  then I got 

sv2v: pattern '{
        b: '1,
        pkg_a: '1,
        default: '0
} has extra named fields ["pkg_a"] that are not in struct packed {
        logic a;
        logic b;
        logic c;
}, within scope b (use -v to get approximate source location)
CallStack (from HasCallStack):
  error, called at src/Convert/Scoper.hs:376:22 in main:Convert.Scoper

Why 'variable a' is replaced as 'function a  in package pkg' ! 
It's ok when I use  'verilator --lint-only test.sv' !

I found this case in PULP's ara source file , see attached file x.zip!
(sv2v -v -E UnbasedUnsized --top=ara_tb_verilator -D VERILATOR -D VLEN=4096 -D NrLanes=4 x.sv)

[x.zip](https://github.com/zachjs/sv2v/files/12708943/x.zip)

zachjs, 2023-11-05:
I apologize for the delay in getting to this. The original issue should be fixed as of a4639fa9ef21fdc8d1a04632a1698fbb5eecf9b3. I also fixed two more issues demonstrated by your larger test case. Please let me know if it works for you.

Thank you very much for filing this! This was quite helpful.

closed 2023-11-16 by forthyen