zachjs/sv2v #256: Issue with $clog2 of a parameter

flaviens, 2023-10-30:
Hi there!
I'm trying to pass [the EL2 core](https://github.com/chipsalliance/Cores-VeeR-EL2) through Yosys. sv2v works without error, but  yosys is then complaining about `$clog2` of a parameter. It seems that the parameter during sv2v, but I'm not very confident about that.

Here is a pickle ready to be consumed by sv2v.
[el2_pickle.zip](https://github.com/zachjs/sv2v/files/13207602/el2_pickle.zip)

```
Generating RTLIL representation for module `$paramod$2bdb596ed0b0d0fd78c36713ac2804f44d511569\el2_lsu_dccm_mem'.
generated/sv2v_out.v:38018: ERROR: Failed to evaluate system function `\$clog2' with non-constant value.        
```

Could you please have a quick look?
I'm at your disposal, may you have any question.
Thank you very much!
Flavien

zachjs, 2023-11-02:
I have an idea what's going on here, but I'll need some time. I intend to take a closer look this weekend.

zachjs, 2023-11-04:
The `$clog2` was actually unrelated. sv2v saw a pattern field name and thought it referred to a localparam defined further down due some extra indirection. This is closely related to #255. I've added logic to distinguish the typenames in struct patterns in a4639fa9ef21fdc8d1a04632a1698fbb5eecf9b3. Please let me know if it works for you. The converted output seems right to me, but I now believe you'll run into an unrelated Yosys limitation: YosysHQ/yosys#668.

This test case was very useful. Thank you very much for filing this issue!

flaviens, 2023-11-05:
Hi Zach, thank you so much for your reactivity! This indeed seems to resolve the problem! :heart: 
As you mentioned, we run into https://github.com/YosysHQ/yosys/issues/668, although for this specific purpose, it is easier to [adapt the source code syntax](https://github.com/chipsalliance/Cores-VeeR-EL2/pull/129).
Thanks!

closed 2023-11-05 by flaviens