zachjs/sv2v #257: package import declaration not work?

YikeZhou, 2023-11-01:
Please consider this example. It comes from [tests/ImportFunction/top.sv](https://github.com/chipsalliance/UHDM-integration-tests/blob/1bd3e0bb9797778570423f44dc848e1d14c6d75f/tests/ImportFunction/top.sv) in the [UHDM-integration-tests](https://github.com/chipsalliance/UHDM-integration-tests) repository.

```systemverilog
package prim_util_pkg;
   function automatic int get_5();
      int result = 5;
      return result;
   endfunction // get_5
endpackage // prim_utilt_pkg
      
package lc_ctrl_state_pkg;
   import prim_util_pkg::get_5;
endpackage // lc_ctrl_state_pkg

module top(output int o);
   assign o = lc_ctrl_state_pkg::get_5();
endmodule; // top
```

When running
```
sv2v top.sv
```
I got
```
sv2v: could not find "get_5" in package "lc_ctrl_state_pkg"
CallStack (from HasCallStack):
  error, called at src/Convert/Package.hs:662:42 in main:Convert.Package
```

Importing `get_5()` inside the module declaration seems to work, though.

```systemverilog
package prim_util_pkg;
   function automatic int get_5();
      int result = 5;
      return result;
   endfunction // get_5
endpackage // prim_utilt_pkg

module top(output int o);
   import prim_util_pkg::get_5;
   assign o = get_5();
endmodule; // top
```

```
$ sv2v top.sv
module top (o);
        output wire signed [31:0] o;
        function automatic signed [31:0] prim_util_pkg_get_5;
                input reg _sv2v_unused;
                reg signed [31:0] result;
                begin
                        result = 5;
                        prim_util_pkg_get_5 = result;
                end
        endfunction
        assign o = prim_util_pkg_get_5(0);
endmodule
```

zachjs, 2023-11-02:
Unfortunately, that test case is invalid. If the test used `export prim_util_pkg::get_5;` in `lc_ctrl_state_pkg`, it would work as expected.

From Section 26.6 of IEEE 1800-2017:
> By default, declarations imported into a package are not visible by way of subsequent imports of that package. Package export declarations allow a package to specify that imported declarations are to be made visible in subsequent imports.

YikeZhou, 2023-11-02:
Thank you so much for guiding me to the proper section. I really appreciate it! In my opinion, sv2v seems to behave quite reasonably.
(In case you haven't already tried it, I just tried the four commercial simulators on edaplayground.com as you suggested in another issue. Half of them accepted the code without error. I suppose encountering such "undefined behaviors" in SystemVerilog is not that uncommon, is it?)

zachjs, 2023-11-05:
> In my opinion, sv2v seems to behave quite reasonably.

I'm glad you agree!

> I just tried the four commercial simulators on edaplayground.com as you suggested in another issue. Half of them accepted the code without error. I suppose encountering such "undefined behaviors" in SystemVerilog is not that uncommon, is it?

Indeed, there are numerous ambiguities within the SystemVerilog specification. However, it's important to distinguish:
- the behaviors that are genuinely ambiguous in the specification; and
- the behaviors that happen to vary across tools, commercial or otherwise.

These two sets have limited overlap. In this case, there is disagreement among tools, but I think the specification is unambiguous.

closed 2023-11-17 by zachjs