zachjs/sv2v #258: Unexpected interface mismatch error when using "modport"

YikeZhou, 2023-11-02:
I'm uncertain if I'm using "modport" incorrectly or if sv2v has extra limitations on its usage.

```systemverilog
interface SimpleInterface;
  logic [15:0] a, b;
  modport oa_ib (output a, input b);
  modport ia_ob (input a, output b);
endinterface

module top;
  logic [15:0] mid_a;
  SimpleInterface interface1();
  mid m(interface1, mid_a);
endmodule

module mid (SimpleInterface.ia_ob mid_int, output logic [15:0] mid_o);
  bottom b(mid_int);
  assign mid_int.b = 16'hdead;
  assign mid_o = mid_int.a;
endmodule

module bottom (SimpleInterface.oa_ib bottom_int);
  assign bottom_int.a = 16'hbeef;
endmodule
``` 

I assume it describes a circuit like this:
[image]

I simulate it with ModelSim. It compiles and works like this:
[image]

However, sv2v would reject this.
```
sv2v: port bottom_int has type SimpleInterface.oa_ib, but the binding top.interface1.ia_ob has type SimpleInterface.ia_ob, within scope top.m (use -v to get approximate source location)
CallStack (from HasCallStack):
  error, called at src/Convert/Scoper.hs:376:22 in main:Convert.Scoper
```

I looked into _25.5 Modports_ in [IEEE 1800-2017](https://ieeexplore.ieee.org/servlet/opac?punumber=8299593) and found that:
> If a port connection specifies a modport list name in both the module instance and module header declaration, then the two modport list names shall be identical.

Does it mean this example is legal SystemVerilog since there is no conflict between `bottom b(mid_int)` and `module bottom (SimpleInterface.oa_ib bottom_int);`? Could there be additional restrictions when using "modport" that I mistakenly ignored?

zachjs, 2023-11-02:
Per the spec referenced, "the two modport list names shall be identical". In this case, the module header uses the modport list name `oa_ib`, while the module instance `b` implicitly uses the modport list name `ia_ob`, as instances of module `mid` are restricted to that view of the interface. Admittedly, I think the spec is ambiguous here. However, of the 4 commercial simulators on edaplayground.com, 2 generate an error, and 1 generates a warning, so sv2v's behavior is in the majority.

This test case can be fixed by changing the module header to:
```SystemVerilog
module mid (SimpleInterface mid_int, output logic [15:0] mid_o);
```

zachjs, 2023-11-04:
I see your +1 above, so I assume you agree with sv2v's behavior here. Please feel free to reopen if necessary.

closed 2023-11-04 by zachjs