zachjs/sv2v #261: `input reg` is not allowed in Verilog

YikeZhou, 2023-11-03:
It appears that sv2v cannot handle the `input reg` in SystemVerilog, which results in invalid Verilog code.

Example input:
```systemverilog
module top(
    input reg [15:0]x,
    input reg [15:0]y,
    output reg [15:0]r
);

endmodule
``` 

Output:
```verilog
module top (
	x,
	y,
	r
);
	input reg [15:0] x;
	input reg [15:0] y;
	output reg [15:0] r;
endmodule
```

zachjs, 2023-11-05:
Thanks for pointing this out! I believe your interpretation of the spec is correct. This is addressed in 2579bc8302522bd058765d4e08d45fc4f959b65b. Please let me know if it works for you!

zachjs, 2023-12-05:
I believe this issue is resolved. Please feel free to reopen it if you're still running into problems.

closed 2023-12-05 by zachjs