zachjs/sv2v #262: Streaming operator is not converted when combined with conditional operator

KatCe, 2023-11-09:
Hello,

In the code example below the first occurrence of the streaming operator (in the assignment to 'out') is not converted to Verilog.

```
module streaming_operator_test
(
  input  logic [31:0] inp,
  input  logic        en,
  output  logic [31:0] out,
  output  logic [31:0] out2
);

always_comb begin
  out = en ? {<<1 {inp}} : 32'b0;
  out2 = {<<1 {inp}};
end

endmodule
```

Sv2v output:

```
module streaming_operator_test (
	inp,
	en,
	out,
	out2
);
	input wire [31:0] inp;
	input wire en;
	output reg [31:0] out;
	output reg [31:0] out2;
	always @(*) begin
		out = (en ? {<< 1{inp}} : 32'b00000000000000000000000000000000);
		begin : sv2v_autoblock_1
			reg [31:0] _sv2v_strm_55E18_inp;
			reg [31:0] _sv2v_strm_55E18_out;
			integer _sv2v_strm_55E18_idx;
			_sv2v_strm_55E18_inp = {inp};
			for (_sv2v_strm_55E18_idx = 0; _sv2v_strm_55E18_idx <= 31; _sv2v_strm_55E18_idx = _sv2v_strm_55E18_idx + 1)
				_sv2v_strm_55E18_out[31 - _sv2v_strm_55E18_idx-:1] = _sv2v_strm_55E18_inp[_sv2v_strm_55E18_idx+:1];
			out2 = _sv2v_strm_55E18_out << 0;
		end
	end
endmodule

```

Tested with commit 2579bc8302522bd058765d4e08d45fc4f959b65b

zachjs, 2023-12-18:
Thank you for filing this issue! sv2v should now support this pattern as of 756dbbb84f637966c114988465ef9164595ff57c. Please let me know if it works for you!

KatCe, 2024-01-08:
Thank you @zachjs for implementing the fix! It works as expected.