zachjs/sv2v #264: fork-join with wait produce a parse error

gggmmm, 2023-11-27:
MRE of [this code](https://github.com/bwitherspoon/rv32cpu/blob/master/rtl/axi4/axi.sv#L135-L138)
```
interface my_if();
    task my_task();
        fork
            wait;
        join
    endtask
endinterface
```
gives
`test.sv:4:13: Parse error: missing expected join`

using `sv2v test.sv -w adjacent`

Seems the wait is interfering somehow. Changing it to eg. $display produces `Warning: Source includes an interface but the output is empty...`

zachjs, 2023-12-03:
Thanks for filing this! I think the reduced example is missing an expression for the wait statement. I have added support for wait statements in e9c01d243454a382b566e2fd1ac4baa5b38c5ca8. Please let me know if it works for you!

zachjs, 2024-01-29:
Because I believe this issue is now resolved, I am closing it. Please reopen it or file a new one if you have further issues!

closed 2024-01-29 by zachjs