zachjs/sv2v #278: Parse error on disable in named block

spth, 2024-02-29:
Using the latest release of sv2v, I get Parse error: missing expected `end on this code:

```
function automatic logic [7:0] clz(logic [15:0] op);
	logic [7:0] count = 16;
	begin: loop
	for (int i = 0; i < 16; i++)
	begin
		if (op[15 - i])
		begin
			count = i;
			//break; // Icarus Verilog 11 doesn't support break.
			disable loop;
		end
	end
	end: loop
	return count;
endfunction
```

zachjs, 2024-04-02:
I believe if you replace the disable with the commented out break, sv2v supports this logic. Does that work for you?

spth, 2024-04-02:
Yes, then the code work with sv2v, but no longer with Icarus 11.

zachjs, 2024-05-14:
I think there are three options here:
1. Add support for disable to sv2v.
2. Add support for break to iverilog.
3. Use break, but pass your input through sv2v before giving it to iverilog.

Does the third option work around your immediate issue?

spth, 2024-05-14:
I've since refactored my design to use neither break nor disable.

Still, both are in the Verilog standard, so the right solution would be to have both 1 (this ticket) and 2 (implemented now, but currently broken https://github.com/steveicarus/iverilog/issues/1016).