zachjs/sv2v #279: File name too long

RHamalainen, 2024-03-22:
Hello,

I have about 1000 SystemVerilog files which I find, list and feed to the `sv2v` with

```
from pathlib import Path
from subprocess import run

if __name__ == "__main__":
    path_root = Path.cwd()
    system_verilog_paths = list(path_root.glob("**/*.sv"))
    # print(len(system_verilog_paths))  # ~1000
    paths_list = [str(path) for path in system_verilog_paths]
    paths_string = " ".join(paths_list)
    run(["sv2v", paths_string])
```

which results

```
sv2v: <files delimited with spaces>: openFile: invalid argument (File name too long)
```

Did I make a mistake or what happened?

zachjs, 2024-04-02:
Can you try passing in each file as a separate argument? Joining them makes it one long file path that both doesn't exist and is longer than supported by the file system here.

zachjs, 2024-04-22:
@RHamalainen Did the suggestion above work for you? Are you still facing an issue with sv2v?

zachjs, 2024-05-10:
@RHamalainen Unless you are to follow up, I'll be closing this issue soon. Please let me know if there's anything I can do to help you!

closed 2024-05-13 by zachjs