zachjs/sv2v #280: regarding system verilog to verilog

sundeep2249, 2024-04-13:
Can you let me know on how to convert system verilog to verilog for yosys

zachjs, 2024-04-14:
What have you tried to this point? sv2v's README includes basic usage instructions. The output is intended to be Yosys-compatible.

sundeep2249, 2024-04-14:
Hi, Does these verilog files works on Intel quartus? I'm getting some errors after I install the stack and then using the make command

zachjs, 2024-04-14:
> Does these verilog files works on Intel quartus?

I don't have access to Quartus, so I can't say for sure. However, I would be interested in fixing issues that affect the portability of sv2v's output. I do know there are other users who use sv2v's output as the input to commercial tools.

> I'm getting some errors after I install the stack and then using the make command

Based on https://discourse.haskell.org/t/cabal-and-llvm-issue/3672/13, the version of the compiler sv2v was using (GHC 8.10.7) doesn't support compiling for Apple Silicon (`aarch64-osx`) without using LLVM, which doesn't come preinstalled on macOS. However, never versions should have native support. I just pushed a changed to upgrade sv2v to a newer (though not the newest) version of the compiler (GHC 9.6.4). Please try running `git pull` and then `make` again.

jrrk2, 2024-04-15:
The dependency on llvm seems to have gone away. There are some failures in the test suite, I don't know if expected or not.

[make_test_discrepancies.txt](https://github.com/zachjs/sv2v/files/14977222/make_test_discrepancies.txt)

jrrk2, 2024-04-15:
With this latest version, sv2v compiles cva6(arcane RISCV processor) correctly with one change:

diff --git a/core/acc_dispatcher.sv b/core/acc_dispatcher.sv
index 8b5998ae..6f9c8fc2 100644
--- a/core/acc_dispatcher.sv
+++ b/core/acc_dispatcher.sv
@@ -67,8 +67,8 @@ module acc_dispatcher
     output logic inval_valid_o,
     output logic [63:0] inval_addr_o,
     // Accelerator interface
-    output acc_req_t acc_req_o,
-    input acc_resp_t acc_resp_i
+    output acc_pkg::accelerator_req_t acc_req_o,
+    input acc_pkg::accelerator_resp_t acc_resp_i
 );
 
   `include "common_cells/registers.svh"

the significance of this is that acc_req_t and acc_resp_t are type parameters and as such get assigned at elaboration time, perhaps sv2v is trying to find the fields before elaboration  (just speculation)

sundeep2249, 2024-04-15:
Hello,

Does sv2v tool converts system verilog files which includes another system verilog file?

zachjs, 2024-04-16:
@jrrk2:
* I'm glad the LLVM dependency is indeed gone. Thanks for confirming!
* As for the tests, because both iverilog and sv2v and its test cases are changing over time, the known-compatible version of iverilog is tracked in `.github/workflows/main.yaml`. I have just upgraded the iverilog version used in CI to their latest commit, fixing some compatibility issues over the past few days. Please pull and see if the tests pass with the new version.
* Can you also provide a complete reproducer for the cva6 conversion issue you're facing? This will help me replicate the issue myself and identify the root cause.

@sundeep2249 sv2v does support `` `include ``. sv2v has pretty good support for most synthesizable and commonly-used language features. I encourage you to try it out and see if it works for you!

jrrk2, 2024-04-23:
I pushed a pull request that demonstrates the problem with type parameters:

https://github.com/zachjs/sv2v/pull/281