zachjs/sv2v #282: How to build on GNU/Linux on ppc64?

spth, 2024-04-25:
~~~~
philipp@nemesis:~/sv2v-0.0.11$ LANG=C make
mkdir -p bin
stack install --install-ghc --local-bin-path bin
Error: [S-1852]
I don't know how to install GHC for (Linux,PPC64), please install manually
make: *** [Makefile:7: sv2v] Error 1
philipp@nemesis:~/sv2v-0.0.11$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 9.4.7
~~~~

zachjs, 2024-04-25:
I likely can't provide much help here because don't have access to a PowerPC machine. However, this build failure isn't specific to sv2v. Any advice you find online for installing a system GHC on PowerPC and setting it up in Stack should be applicable here too.

If you do get further along in the build, please share some pointers! There may be other users that would benefit.

Bodigrim, 2024-04-30:
@spth could you please try `cabal unpack sv2v-0.0.11 && cd sv2v-0.0.11 && cabal install --allow-newer -w ghc-9.4`? This is assuming you `apt-get install cabal-install ghc`.

spth, 2024-05-07:
Well, I don't have any experience with Haskell (I'm a C programmer). I've tried this variant of the above: `cabal unpack sv2v-0.0.12 && cd sv2v-0.0.12 && cabal install --allow-newer -w ghc-9.4.7` but it fails before it gets to sv2v:

```
Failed to build os-string-2.0.2.1.
Build log (
/home/philipp/.cabal/logs/ghc-9.4.7/os-string-2.0.2.1-06e42e20ec7bc311f28ef3125b4ae8133648473a05a09693d632a831139a5ff0.log
):
Configuring library for os-string-2.0.2.1..
Preprocessing library for os-string-2.0.2.1..
Building library for os-string-2.0.2.1..
[ 1 of 10] Compiling System.OsString.Data.ByteString.Short.Internal ( System/OsString/Data/ByteString/Short/Internal.hs, dist/build/System/OsString/Data/ByteString/Short/Internal.o, dist/build/System/OsString/Data/ByteString/Short/Internal.dyn_o )

System/OsString/Data/ByteString/Short/Internal.hs:314:15: error:
    • Couldn't match type ‘Word#’ with ‘Word16#’
      Expected: Word16# -> Word16#
        Actual: Word# -> Word#
    • In the expression: byteSwap16#
      In an equation for ‘word16ToLE#’: word16ToLE# = byteSwap16#
    |
314 | word16ToLE# = byteSwap16#
    |               ^^^^^^^^^^^
Error: cabal: Failed to build os-string-2.0.2.1 (which is required by exe:sv2v
from sv2v-0.0.12). See the build log above for details.
```

P.S.: To me personally, the ppc64 is less important than amr64 now: My synthesis workflow is sv2v -> yosys, and yosys depends on abc, which has a bugs preventing it from working on ppc64.

Bodigrim, 2024-05-09:
@spth thanks for the bug report. While Haskell in principle works on any system with LLVM, it lacks testing on big-endian arches unfortunately. This is now fixed in https://github.com/haskell/os-string/pull/14, could you please `cabal update` and try again?

spth, 2024-05-14:
It looks like this might have worked:

~~~~
philipp@nemesis:/tmp/test$ cabal unpack sv2v-0.0.12 && cd sv2v-0.0.12 && cabal install --allow-newer -w ghc-9.4.7
Unpacking to sv2v-0.0.12/
Wrote tarball sdist to
/tmp/test/sv2v-0.0.12/dist-newstyle/sdist/sv2v-0.0.12.tar.gz
Resolving dependencies...
Build profile: -w ghc-9.4.7 -O1
In order, the following will be built (use -v for more details):
 - os-string-2.0.2.2 (lib) (requires download & build)
 - vector-0.13.1.0 (lib) (requires build)
 - hashable-1.4.4.0 (lib) (requires build)
 - sv2v-0.0.12 (exe:sv2v) (requires build)
Downloading  os-string-2.0.2.2
Starting     vector-0.13.1.0 (lib)
Downloaded   os-string-2.0.2.2
Starting     os-string-2.0.2.2 (lib)
Building     vector-0.13.1.0 (lib)
Building     os-string-2.0.2.2 (lib)
Installing   os-string-2.0.2.2 (lib)
Completed    os-string-2.0.2.2 (lib)
Starting     hashable-1.4.4.0 (lib)
Building     hashable-1.4.4.0 (lib)
Installing   hashable-1.4.4.0 (lib)
Completed    hashable-1.4.4.0 (lib)
Installing   vector-0.13.1.0 (lib)
Completed    vector-0.13.1.0 (lib)
Starting     sv2v-0.0.12 (exe:sv2v)
Building     sv2v-0.0.12 (exe:sv2v)
Installing   sv2v-0.0.12 (exe:sv2v)
Completed    sv2v-0.0.12 (exe:sv2v)
Symlinking 'sv2v' to '/home/philipp/.cabal/bin/sv2v'
philipp@nemesis:/tmp/test/sv2v-0.0.12$
~~~~

How would I run the tests here?

zachjs, 2024-05-14:
Please try the following:

- Clone the full sv2v repo. The Hackage tarball doesn't include the tests, as users don't generally need to run them.
- Make your out-of-band build of sv2v available at `<clone>/bin/sv2v`. The binary is statically linked.
- Run `make test`. This assumes you have `iverilog` and `python3` in your `PATH`.

Let me know if this works for you!

spth, 2024-05-16:
I see "5 test suite(s) failed".
[log.gz](https://github.com/zachjs/sv2v/files/15334594/log.gz)

zachjs, 2024-05-16:
@spth Those test failures all appear to be due to an an outdated version of iverilog, as in https://github.com/zachjs/sv2v/issues/283#issuecomment-2098264126.

It does look like you have a working build of sv2v on PPC. Thanks for the help, @Bodigrim!

zachjs, 2024-06-15:
I am closing this and #283 as these build issues appear to be resolved. Please feel free to reopen if not. Thank you @spth, @Bodigrim, and @newAM for your help!

closed 2024-06-15 by zachjs