zachjs/sv2v #283: How to build on GNU/Linux on arm64?

spth, 2024-04-30:
I'm trying to build sv2v on Debian GNU/Linux on arm64. First I ran into:

~~~~
philipp@alecto:/tmp/sv2v-0.0.11$ make
mkdir -p bin
stack install --install-ghc --local-bin-path bin

<no location info>: error:
    Warning: Couldn't figure out LLVM version!
             Make sure you have installed LLVM between [9 and 13)
[1 of 2] Compiling Main             ( /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.hs, /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.o )

<no location info>: error:
    Warning: Couldn't figure out LLVM version!
             Make sure you have installed LLVM between [9 and 13)
ghc: could not execute: opt

Error: [S-6374]
       While building simple Setup.hs (scroll up to its section to see the error) using:
       /home/philipp/.stack/programs/aarch64-linux/ghc-tinfo6-8.10.7/bin/ghc-8.10.7 -rtsopts -threaded -clear-package-db -global-package-db -hide-all-packages -package base -main-is StackSetupShim.mainOverride -package Cabal-3.2.1.0 /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.hs /home/philipp/.stack/setup-exe-src/setup-shim-sDt42OhJ.hs -o /home/philipp/.stack/setup-exe-cache/aarch64-linux-tinfo6/tmp-Cabal-simple_sDt42OhJ_3.2.1.0_ghc-8.10.7
       Process exited with code: ExitFailure 1 
make: *** [Makefile:7: sv2v] Fehler 1
~~~~

Which I got around via "apt-get -t testing install llvm-dev". Unfortunately, I don't get much further:

~~~~
philipp@alecto:/tmp/sv2v-0.0.11$ make
mkdir -p bin
stack install --install-ghc --local-bin-path bin
[1 of 2] Compiling Main             ( /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.hs, /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.o )
The `opt -passname` syntax for the new pass manager is not supported, please use `opt -passes=<pipeline>` (or the `-p` alias for a more concise version).
See https://llvm.org/docs/NewPassManager.html#invoking-opt for more details on the pass pipeline syntax.

`opt' failed in phase `LLVM Optimiser'. (Exit code: 1)

Error: [S-6374]
       While building simple Setup.hs (scroll up to its section to see the error) using:
       /home/philipp/.stack/programs/aarch64-linux/ghc-tinfo6-8.10.7/bin/ghc-8.10.7 -rtsopts -threaded -clear-package-db -global-package-db -hide-all-packages -package base -main-is StackSetupShim.mainOverride -package Cabal-3.2.1.0 /home/philipp/.stack/setup-exe-src/setup-sDt42OhJ.hs /home/philipp/.stack/setup-exe-src/setup-shim-sDt42OhJ.hs -o /home/philipp/.stack/setup-exe-cache/aarch64-linux-tinfo6/tmp-Cabal-simple_sDt42OhJ_3.2.1.0_ghc-8.10.7
       Process exited with code: ExitFailure 1 
make: *** [Makefile:7: sv2v] Fehler 1
~~~~

zachjs, 2024-04-30:
What version of LLVM are you using?

spth, 2024-04-30:
~~~~
root@alecto:~# LANG=C apt list llvm-dev
Listing... Done
llvm-dev/testing,now 1:16.0-57 arm64 [installed]
~~~~

Bodigrim, 2024-04-30:
@spth similar to #282, I'd suggest to try `cabal unpack sv2v-0.0.11 && cd sv2v-0.0.11 && cabal install --allow-newer -w ghc-9.4`. This is assuming you have GHC 9.4 from a system package manager.

zachjs, 2024-04-30:
I recommend using the latest commit of sv2v, which already builds against GHC 9.4.

It looks like you're using LLVM 16. Per https://www.haskell.org/ghc/download_ghc_9_4_7.html, GHC 9.4 is only compatible with LLVM 10 through 14. However, I'm surprised GHC doesn't notice this incompatibility itself, so I might be missing something.

sv2v can't yet build against GHC 9.6, which supports LLVM 15. I'll work on making sv2v compatible with newer versions of GHC.

Bodigrim, 2024-04-30:
LLVM version is irrelevant on aarch64 starting from GHC 9.2+, because it got a native code generator.

zachjs, 2024-05-05:
@spth To clarify the above, although sv2v now builds by default against GHC 9.6.4, it doesn't actually require such a new version. As of 988f76b92bcbb081adc2d46f7a47526e6529b2a6, sv2v builds against GHC as old as 8.6.4 (released over 5 years ago) and as new as 9.8.2 (released in February). Building against these versions doesn't require _any_ change to sv2v's source; any functional Stack or GHC+Cabal should work! Please let me know if you're still facing issues, or what you did to get this working.

Please also let me know if generating a newer release would help. v0.0.11 is from last summer.

@newAM A while back (in #186) you looked at sv2v's setup in NixOS (which I don't personally maintain). I noticed sv2v is currently marked "broken", but I can't find the cause of the failure in Hydra. I know fairly little about nix on the maintenance side. Can you give me some pointers? I'm guessing a working nix build would improve support for other platforms.

newAM, 2024-05-07:
The last working build is in release-22.11, and it still builds today with `nix-build -A haskellPackages.sv2v --argstr system aarch64-linux`

I have not used it in a while so I didn't notice it was broken in nixpkgs.
I tried to build `sv2v` with the latest master, got this log:

```
Configuring sv2v-0.0.11...
CallStack (from HasCallStack):
  withMetadata, called at libraries/Cabal/Cabal/src/Distribution/Simple/Utils.hs:368:14 in Cabal-3.10.3.0:Distribution.Simple.Utils
Error: Setup: Encountered missing or private dependencies:
base <4.15, hashable <1.4, mtl <2.3, vector <0.13
```

I have never used haskell package managers, but I think this means a dependency update would fix the build failure?

zachjs, 2024-05-07:
@newAM Thanks for looking! I just created the v0.0.12 release [on GitHub](https://github.com/zachjs/sv2v/releases/tag/v0.0.12) and [on Hackage](https://hackage.haskell.org/package/sv2v-0.0.12). The Hackage build doesn't like the included bounds (unfortunately `stack sdist` and `stack upload` appear to have different defaults). I'm curious if the nix build will tolerate them.

spth, 2024-05-07:
I've just tried the 0.0.12 release on my Debian GNU/Linux arm64 system. I can now build via "make". But I see some failures on "make test": `4 test suite(s) failed`.

zachjs, 2024-05-07:
@spth I’m glad to hear you got this building on your system! Those test failures are due to the version of iverilog installed. Because iverilog’s behavior changes from time to time (typically for the better!), sv2v’s main CI workflow freezes the version of iverilog used. I believe the version I’m using is close to the latest master. Note that these issues are only with the “test environment”; sv2v doesn’t use iverilog to perform the conversion.

newAM, 2024-05-09:
I tried out nixpkgs with 0.0.12, both `x86_64-linux` and `aarch64-linux` work.  The `haskellPackages` set in nixpkgs gets updated automatically (e.g. https://github.com/NixOS/nixpkgs/pull/307204).  It will take a few weeks for it to propagate into nixpkgs, but I'm glad it's going to get fixed soon!  Thanks for the update.

zachjs, 2024-05-14:
As far as I know, the only outstanding item in this issue is ensuring nix regains a working build of sv2v. Please let me know if I'm missing something!

newAM, 2024-05-20:
sv2v was updated in nixpkgs (https://github.com/NixOS/nixpkgs/pull/311940), but it still has the broken flag, I have opened a PR to remove this flag in nix: https://github.com/NixOS/nixpkgs/pull/313171

newAM, 2024-05-26:
`sv2v` now builds on aarch64-linux in nixpkgs (`nixpkgs-unstable`/`nixos-unstable`) again.

zachjs, 2024-06-15:
@newAM Thank you for fixing the nix build of sv2v! https://search.nixos.org/packages?type=packages&query=sv2v

I am closing this and #282 as these build issues appear to be resolved. Please feel free to reopen if not. Thank you @spth, @Bodigrim, and @newAM for your help!

closed 2024-06-15 by zachjs