zachjs/sv2v #284: Build fails on MacOS

zaun, 2024-04-30:
Any idea what's going on?

```
make
mkdir -p bin
stack install --install-ghc --local-bin-path bin
hashable> configure
hashable> Configuring hashable-1.4.3.0...
hashable> build with ghc-9.4.8
hashable> Preprocessing library for hashable-1.4.3.0..
hashable> Building library for hashable-1.4.3.0..
hashable> [1 of 7] Compiling Data.Hashable.Imports
hashable> [2 of 7] Compiling Data.Hashable.LowLevel
hashable>   
hashable> /private/var/folders/2n/9d_8f0dd7tj6bwwx3tmxrgn80000gn/T/stack-f44f2274842add8c/hashable-1.4.3.0/In file included from /var/folders/2n/9d_8f0dd7tj6bwwx3tmxrgn80000gn/T/ghc20852_0/ghc_10.c:4:0: error:
hashable>     
hashable>   
hashable> /Users/justinzaun/.stack/programs/aarch64-osx/ghc-9.4.8/lib/ghc-9.4.8/lib/../lib/aarch64-osx-ghc-9.4.8/rts-1.0.2/include/rts/ghc_ffi.h:28:10: error:
hashable>      fatal error: 'ffi.h' file not found
hashable>    |
hashable> 28 | #include "ffi.h"
hashable>    |          ^
hashable> #include "ffi.h"
hashable>          ^~~~~~~
hashable> 1 error generated.
hashable>   
hashable> <no location info>: error:
hashable>     `gcc' failed in phase `C Compiler'. (Exit code: 1)
Progress 1/2

Error: [S-7282]
       Stack failed to execute the build plan.
       
       While executing the build plan, Stack encountered the error:
       
       [S-7011]
       While building package hashable-1.4.3.0 (scroll up to its section to see the error) using:
       /Users/justinzaun/.stack/setup-exe-cache/aarch64-osx/Cabal-simple_6HauvNHV_3.8.1.0_ghc-9.4.8 --verbose=1 --builddir=.stack-work/dist/aarch64-osx/ghc-9.4.8 build --ghc-options " -fdiagnostics-color=always"
       Process exited with code: ExitFailure 1 
make: *** [sv2v] Error 1
```

I do have libffi installed:

```
brew install libffi
Warning: libffi 3.4.6 is already installed and up-to-date.
```

zachjs, 2024-04-30:
This build failure isn't specific to sv2v. Unfortunately, I don't currently have access to an Apple Silicon Mac to debug builds on.

Do you think https://github.com/haskell/ghcup-hs/issues/814#issuecomment-1485045426 is relevant?

zaun, 2024-05-01:
`xcode-select --install` did fix the issue. It's funny, no other builds ive been doing required that to work. anyway, thanks.

closed 2024-05-01 by zaun