zachjs/sv2v #285: SV2V QA test suite

chili-chips-ba, 2024-06-14:
While trying SV2V, we quickly realized how [good](https://github.com/chili-chips-ba/openCologne/issues/3#issuecomment-2161988228) it was compared to both _Synlig_ and _Yosys-Slang_.

Yet, having run into some here-and-there issues, we believe that the `SV2V` project would benefit from adding our two SV RTL repos to its regression tests:
- [openCam-SI](https://github.com/chili-chips-ba/openeye-CamSI/blob/main/1.hw/top.filelist)
- [opl3_fpga](https://github.com/chili-chips-ba/openCologne/blob/main/4.Advanced--4--Yamaha-OPL3-FM-Synth/1.hw/rtl.filelist)

zachjs, 2024-06-15:
Although sv2v has extensive test coverage (you can check its code coverage with `make coverage`), this repository doesn't include regression tests, per se. sv-tests already covers a handful of large open source SystemVerilog projects. My gut feeling is that it would be best for the community to focus our regression testing efforts into the sv-tests project. What do you think?

chili-chips-ba, 2024-06-15:
While we are not sure about the internal workings of sv2v QA process, we were under the impression that you were taking only [occasional and cursory look](https://github.com/chili-chips-ba/openCologne/issues/3#issuecomment-2156807368) at sv-tests 
> "... indeed sv2v has hundreds of its own tests, generally focusing on just a few features at a time. I do use the sv-tests dashboard from time to time to check in on its assessment of sv2v..."

We also agree that moving the focus of your QA process to sv-tests is a worthy idea. However, to get the most out of it, the sv-tests would need to be divided into two sub-categories:
- for RTL and Synth
- for Sim and DV, perhaps even with UVM sub-sub-category

or provide a mechanism for marking selected SV features (along with tests that validate them) as NA. Those should then show greyed out and not contribute to your total score. We assume that all that the sv2v cares about at the moment is the RTL/Synth subset of the language.

In the meantime, the life goes on, and your private library of test designs may need to grow with it 😉...

chili-chips-ba, 2024-06-15:
@hzeller anything you can do about dividing _sv-tests_ into two sub-categories (1) RTL/Synth; (2) DV/Sim, possible with UVM as a sub-sub-category?!

sifferman, 2024-07-25:
@chili-chips-ba That is how sv-tests currently works for Yosys and Synlig: https://github.com/chipsalliance/sv-tests/pull/5139. The functionality can be easily extended to sv2v

zachjs, 2024-09-01:
Thank you for starting this discussion! I don't think there are outstanding action items for sv2v, as sv-tests is better suited for these sorts of large and widely-applicable test cases. I am closing this issue.

closed 2024-09-01 by zachjs