zachjs/sv2v #287: Non-exhaustive patterns in Part _ _ Module _ name _ _  with --write=DIR option

tarik-ibrahimovic, 2024-06-15:
Using the -write=DIR option the mentioned error arises, while --write=adjacent passes. Log:
`sv2v --write=./ /home/tibrahimovic/projects/2.Simple--1/3.build/../1.hw/csr_pkg.sv /home/tibrahimovic/projects/2.Simple--1/3.build/../1.hw/csr_if.sv /home/tibrahimovic/projects/2.Simple--1/3.build/../1.hw/uart.WITHOUT-FIFO.sv /home/tibrahimovic/projects/2.Simple--1/3.build/../1.hw/psram.sv /home/tibrahimovic/projects/2.Simple--1/3.build/../1.hw/top.sv 
sv2v: src/sv2v.hs:69:9-48: Non-exhaustive patterns in Part _ _ Module _ name _ _`

zachjs, 2024-06-15:
While directly fixing this particular "Non-exhaustive patterns" would be straightforward, I'm concerned that this failure is indicative of some other problem elsewhere in sv2v, e.g., that something isn't being properly converted. Could you please share the input files and/or the output you get from `--write=adjacent`?

tarik-ibrahimovic, 2024-06-15:
I did RTL simulation before and after conversion (using both Verilator and Icarus), searching for an another issue, it would seem that conversion went fine.  All input files are available [here](https://github.com/chili-chips-ba/openCologne/tree/main/2.Simple--1--PSRAM/1.hw).

chili-chips-ba, 2024-06-15:
`"... yosys synthesis (probably) quietly fails in the next step..."` sounds scary -- We recommend opening a Yosys ticket for that issue.

zachjs, 2024-06-16:
Thanks for reporting this issue and sharing your reproducer! I believe it should be fixed as of 6eda946f5739e316f0a71c8cb4761653dffb5e5a. Can you let me know if it's working for you?

tarik-ibrahimovic, 2024-06-17:
Yes, this seems to be working now. Thank you, I am now closing this issue.

closed 2024-06-17 by tarik-ibrahimovic