zachjs/sv2v #288: [Feature Request] Copy comments over

Willyarma, 2024-06-18:
Would it possible to have the ability to have all the comments copied over from the source file?

zachjs, 2024-06-20:
sv2v's internal representation wasn't necessarily designed with this feature in mind (it's better suited for the conversion logic). The possible ways of supporting this feature I have considered so far would be fairly hacky and wouldn't necessarily cover all comments or put them all in the "right" place in the output.

You may find the existing `-v`/`--verbose` mode useful. It can also give you an impression for the existing level of comment granularity.

Can you please describe your intended use case for this feature? What do you intend to do with this output, and how will it fit into your flow?

zachjs, 2024-07-01:
@Willyarma Have you had the chance to look at my response above?

Willyarma, 2024-07-01:
It was just an observation.

The issues with it are:
1. Usually at the start there is copyright and licensing information and the authors credits.
2. The comments throughout the file could help in matching up the input and output to aid in finding bugs and generally makes it more readable.

My feeling is that so long as the converted output is only used for synthesis or simulation and not used as a source to work on or distribute and all modifications are done to the original sv then I suppose it's not a huge issue.

zachjs, 2024-09-01:
> I have not tried the -v option, apologies if this already does what I described.

Indeed `--verbose` already accomplishes at least part of (2) in your list (specifically, "matching up the input and output").

> My feeling is that so long as the converted output is only used for synthesis or simulation and not used as a source to work on or distribute and all modifications are done to the original sv then I suppose it's not a huge issue.

Given the likely high implementation cost and the apparently limited user benefit, I am closing this as "not planned" for now. Please feel free to reopen or comment if these circumstances change!

closed 2024-09-01 by zachjs