zachjs/sv2v #290: Missing "wire" keyword

ldoolitt, 2024-07-31:
With this 94-line test case
[test3.txt](https://github.com/user-attachments/files/16445114/test3.txt) (cut back from 2464 lines of production code) a couple of signals get declared as "signed" instead of "wire signed".  That makes the sv2v output at least incompatible with iverilog.
```
$ sv2v test3.sv > foo.v
$ iverilog -Wall foo.v
foo.v:77: syntax error
foo.v:77: error: invalid module item.
foo.v:78: syntax error
foo.v:78: error: invalid module item.
$ sed -e 's/      signed /        wire signed /' < foo.v > foo_fixed.v # the first character of the regex should be a tab 
$ iverilog -Wall foo_fixed.v 
$
```
I suspect this test case can be cut back even more.  I'll leave that job to someone who understands SystemVerilog better than me.

zachjs, 2024-08-05:
Thank you for filing an issue with such a nice reduced test case! This should be fixed in 1b2734324e63059cef4ab71543f34ec6a6bb59ec. Please let me know if It works for you!

ldoolitt, 2024-08-05:
Yes, it works now.  Thanks!

closed 2024-08-05 by ldoolitt