zachjs/sv2v #291: bit fill literal '0 does not expand (sv2v) properly

ljmalone123, 2024-09-18:
Hi, SV2V is great, I believe there is one small issue. When a bit fill literal such as '0 is converted from sv to v , the number of bits is not properly inferred for the signal declaration.  Thanks Jarrett

--- sv2v input ---
logic [31:0] pc;
always_ff @(posedge i_clk or negedge i_resetb)
    if      (!i_resetb     )     pc <= '0;

--- sv2v  output ---
reg [31:0] pc;
always @(posedge i_clk or negedge i_resetb)
		if (!i_resetb)
			pc <= 1'sb0;

---------------------------------------------------
I believe this should be pc <= 32'b0;
---------------------------------------------------
command used : sv2v fname.sv >> fname.v
---------------------------------------------------

**FULL SV2 Input Module** 
module test1();
logic [31:0] pc;
logic        i_resetb;
logic        i_clk;
initial      i_resetb = 0;
initial      i_clk = 0;

always_ff @(posedge i_clk or negedge i_resetb)
    if      (!i_resetb     )     pc <= '0;
    else                             pc <= pc + 32'd4;
endmodule

**FULL SV2V Output Module** 

module test1;
	reg [31:0] pc;
	reg i_resetb;
	reg i_clk;
	initial i_resetb = 0;
	initial i_clk = 0;
	always @(posedge i_clk or negedge i_resetb)
		if (!i_resetb)
			pc <= 1'sb0;
		else
			pc <= pc + 32'd4;
endmodule

---------------------------------------------------
Verilator lint:
verilator --lint-only fname.v

Verilator output:
%Warning-WIDTHEXPAND: fname.v:476:7: Operator ASSIGNDLY expects 32 bits on the Assign RHS, but Assign RHS's CONST '1'sh0' generates 1 bits.
                                  
  476 |    pc <= 1'sb0;
---------------------------------------------------

zachjs, 2024-09-18:
Thanks for filing this issue! This behavior is an intentional simplification for simple assignments to the value of an unbased unsized literal. Although Verilator outputs a stylistic linting warning, the behavior is identical in this context. Are you encountering a behavioral issue?

ljmalone123, 2024-09-19:
Hi, thanks for quick reply. I did not see any behavioral issue with either pc <= 1'sb0 ('0) or with pc <= 1'sb1  ('1).  it sounds like this is the intended translation, so it certainly works for me.  sv2v is a great tool, thanks!  I'll close this out.

closed 2024-09-19 by ljmalone123