zachjs/sv2v #292: add enum attributes to enum values

rroohhh, 2024-10-11:
`yosys` and for example `amaranth` add `enum_base_type` and `enum_value_{value}` to variables that used to be enums a la 
```
(* enum_base_type = "Port" *)
(* enum_value_000 = "Local" *)
(* enum_value_001 = "North" *)
(* enum_value_010 = "South" *)
(* enum_value_011 = "East" *)
(* enum_value_100 = "West" *)
(* src = "/data/projects/amaranth/amaranth/lib/stream.py:48" *)
wire [2:0] result__payload;
```

zachjs, 2024-10-21:
Thanks for filing this issue! Are you looking to have these enum attributes for debugging, or do they result in some behavioral difference downstream?

rroohhh, 2024-10-21:
Only for debugging, mainly to get the names in the waveform viewer.