zachjs/sv2v #294: Breakage caused by attributes

ldoolitt, 2024-10-23:
I have about 1600 lines of SystemVerilog that I'd like to push through sv2v,
but it turns out they use attributes in a way that sv2v doesn't (yet) handle correctly.
See attached 45 lines of SystemVerilog, and a shell script that exercises it in various configurations to demonstrate the bug.

Sorry about the names of the attachments.  Just following gitlab rules.
Of course they should be renamed to oscope_test.sv and doit.
[oscope_test.txt](https://github.com/user-attachments/files/17498249/oscope_test.txt)
[doit.txt](https://github.com/user-attachments/files/17498255/doit.txt)

Output of `sh doit`:
```
+ sv2v --version
sv2v v0.0.12-17-g5374679
+ echo with attributes: fails
with attributes: fails
+ sv2v -DSIMULATE -DATTR1=(* yellow *) -DATTR2=(* blue *) oscope_test.sv
sv2v: could not find decl dir of periph_config among fromList [("zif_cfg",input)]
CallStack (from HasCallStack):
  error, called at src/Convert/Interface.hs:706:28 in main:Convert.Interface
+ echo 1
1
+ wc test1.v
0 0 0 test1.v
+ echo disable one attribute: half succeeeds
disable one attribute: half succeeeds
+ sv2v -DSIMULATE -DATTR1= -DATTR2=(* blue *) oscope_test.sv
+ echo 0
0
+ wc test2.v
  43  120 1274 test2.v
+ iverilog test2.v
test2.v:28: syntax error
test2.v:30: error: invalid module item.
test2.v:37: syntax error
test2.v:39: error: invalid module item.
+ echo disable both attributes: copacetic
disable both attributes: copacetic
+ sv2v -DSIMULATE -DATTR1= -DATTR2= oscope_test.sv
+ echo 0
0
+ wc test3.v
  41  115 1229 test3.v
+ iverilog test3.v -o /dev/null
```

zachjs, 2024-10-27:
Thank you for reporting this issue! It should be fixed as of 24ab7aee24a0475cb83d6ed4dc55924d11b33b0a. Please let me know if it works for you!

ldoolitt, 2024-10-27:
Nice!  That fixed one of the two bugs.  Sorry for combining them into one issue, but I ran into them both with the same code base, and they both involve attributes.

Now sv2v creates test1.v, but that file is not proper Verilog, at least not according to iverilog.  It suffers from the same problem as test2 above.  Focus on the transcript where it says
```
+ iverilog test2.v
test2.v:28: syntax error
test2.v:30: error: invalid module item.
test2.v:37: syntax error
test2.v:39: error: invalid module item.
```
sv2v has tried to nest generate statements.

zachjs, 2024-10-27:
My bad! Indeed this output isn't right:
```
(* blue *) generate
	if (1) begin : digitizer_config
		wire [31:0] periph_config;
		assign oscope_test.zif_cfg.U2_pdwn = periph_config[1];
		assign oscope_test.zif_cfg.U3_pdwn = periph_config[1];
		assign oscope_test.zif_cfg.U33U1_pwr_en = periph_config[0];
	end
	assign digitizer_config.periph_config = digitizer_config_periph_config;
endgenerate
```
but I'm interested to know what you'd like sv2v to do with the `(* blue *)` attribute after inlining in this case. What do you think?

ldoolitt, 2024-10-27:
I'm pleased you managed to attach the `(* yellow *)` attribute to something constructive.
If that's not possible with `(* blue *)`, it's certainly better to drop it than to emit invalid code.

zachjs, 2024-10-27:
> it's certainly better to drop it than to emit invalid code.

That makes sense to me! Please give 7808819c48c167978aeb5ef34c6e5ed416e90875 a shot.

ldoolitt, 2024-10-27:
Yes, that worked, and I'll close this issue. Thanks for your prompt (and successful) attention!

I messed around with the output Verilog, and sure couldn't find any place for a `(* blue *)` attribute.
I'll go back to my 1600 lines of actual source, and figure out what, if anything, breaks by not having the attribute propagated.
So I'll reserve the right to file a new issue later.  I'd make sure to have some constructive ideas for how it could be rendered in the Verilog output.

closed 2024-10-27 by ldoolitt