zachjs/sv2v #296: Problem with localparam in package

fsiegle, 2024-11-27:
Hi,

First of all, many thanks for your great work!

I have a strange issue, which only occurs in one specific file.

[sv2v_test.zip](https://github.com/user-attachments/files/17936523/sv2v_test.zip)

Running sv2v on this dummy example results in:

```
sv2v: field 'interconn_pkg_MAX_BURST_WIDTH' not found in struct packed {
        logic [31:0] addr;
        logic read;
        logic write;
        logic [31:0] write_data;
        logic [3:0] byte_enable;
        logic [mst_in.interconn_pkg_MAX_BURST_WIDTH:0] burstcount;
}, in expression mst_in.interconn_pkg_MAX_BURST_WIDTH, within scope mst_in, near mst_in.sv:12:5
CallStack (from HasCallStack):
  error, called at src/Convert/Scoper.hs:402:22 in main:Convert.Scoper
```

If you comment out `input  mst_slv_t mst_in` in mst_in.sv it works, so it does not have a problem to find MAX_BURST_WIDTH for `output mst_slv_t slv_out[NO_SLV]`. 

Best regards,
Felix

zachjs, 2024-12-09:
Thank you for reporting this issue! I pushed a fix in c56a91b2909557aa684efc5ff1c68cdcfe3820d3. Please let me know if it works for you!

fsiegle, 2024-12-09:
Thank you! It works on the example code but unfortunately not on the original source file, but the problem seems to be the same or at least related.  I found another minimal example:

[sv2v_test.zip](https://github.com/user-attachments/files/18058964/sv2v_test.zip)

Output:

```
sv2v: field 'NO_SLV' not found in struct packed {
        logic [31:0] addr;
        logic read;
        logic write;
        logic [31:0] write_data;
        logic [3:0] byte_enable;
        logic [mst_in.interconn_pkg_MAX_BURST_WIDTH:0] burstcount;
}, in expression mst_in.mst_in.NO_SLV, within scope mst_in.<unnamed_block>.<unnamed_block>.<unnamed_block>, near mst_in.sv:34:11
CallStack (from HasCallStack):
  error, called at src/Convert/Scoper.hs:402:22 in main:Convert.Scoper
```

zachjs, 2024-12-09:
These minimal reproducers are very helpful! In this case, though, I'm very interested in seeing the full complexity of the `mst_in` module. Could you please upload the full example?

fsiegle, 2024-12-10:
Here with the full module included:

[sv2v_test.zip](https://github.com/user-attachments/files/18078316/sv2v_test.zip)

zachjs, 2024-12-11:
Thank you for sharing the complete example! I just pushed a couple more patches, 1c13bcf5571107b74ccd86c3303250344622f08b and 5a636724d783edd403ac4618d6cc4def72e39cbc. Please let me know if they work for you!

fsiegle, 2024-12-11:
Amazing, it does work! (and also for the overall design). Thanks for the quick fixes.

closed 2024-12-11 by fsiegle