zachjs/sv2v #297: Parsing error

miquelt9, 2024-12-02:
Hi!

I have tried to convert the following file into verilog:

```
module bus_transfer_invert(in, invert, out);
    parameter N=8;
    input [N-1:0] in;
    input invert;
    output [N-1:0] out;
    
    xor X[N-1:0] (out, in, invert);
endmodule
```
It returns the following error:
program_x.v:7:10: Parse error: unexpected token '[' (Sym_brack_l)

However Icarus Verilog seems to be fine with it.
For more context, my goal is to read it in Yosys, and I have already tried the synlig plugin but it doesn't work. Any suggestions are kindly accepted.

zachjs, 2024-12-12:
Thank you for filing this issue! I've added support for gate arrays and multidimensional gate arrays in 12618d541ebda97d69e2a6b511af48a9a967660d. Please let me know if it works for you!

miquelt9, 2024-12-12:
Thank you Zachary, it did work for me. 
I have found that my problem is related however to Yosys (where an [issue](https://github.com/YosysHQ/yosys/issues/4785) has been opened as well).
Thank you again in any case.

closed 2024-12-12 by miquelt9