zachjs/sv2v #301: generates wrong sized initializations on its own generated localparams

pbreuer, 2024-12-21:
I'm seeing a lot of:

`        // removed localparam type sv2v_uu_icache_SelHPTW
        localparam [0:0] sv2v_uu_icache_ext_SelHPTW_0 = 1'sb0;
        localparam sv2v_uu_icache_WORDLEN = 32;
        // removed localparam type sv2v_uu_icache_ByteMask
        localparam [3:0] sv2v_uu_icache_ext_ByteMask_0 = 1'sb0;
        localparam sv2v_uu_icache_LOGBWPL = AHBWLOGBWPL;
        // removed localparam type sv2v_uu_icache_BeatCount
        localparam [sv2v_uu_icache_LOGBWPL - 1:0] sv2v_uu_icache_ext_BeatCount_0 = 1'sb0;
        // removed localparam type sv2v_uu_icache_SelBusBeat
        localparam [0:0] sv2v_uu_icache_ext_SelBusBeat_0 = 1'sb0;
        // removed localparam type sv2v_uu_icache_WriteData
        localparam [31:0] sv2v_uu_icache_ext_WriteData_0 = 1'sb0;
        // removed localparam type sv2v_uu_icache_FlushCache
        localparam [0:0] sv2v_uu_icache_ext_FlushCache_0 = 1'sb0;
        // removed localparam type sv2v_uu_icache_CMOpM
        localparam [3:0] sv2v_uu_icache_ext_CMOpM_0 = 1'sb0;
`

I don't know exactly what is going on here but the initialization constants are the wrong size for some of these declarations.

Regards

PTB

zachjs, 2024-12-22:
Thank you for filing this issue! For these localparams, sv2v relies on the sign and size extension of assignment like contexts defined in the Verilog standard to simplify its conversion. That is,
```Verilog
localparam [3:0] sv2v_uu_icache_ext_ByteMask_0 = 1'sb0;
```
is behaviorally equivalent to
```Verilog
localparam [3:0] sv2v_uu_icache_ext_ByteMask_0 = 3'b000;
```
Do you agree? If not, what am I missing?

pbreuer, 2024-12-22:
Hi ...

On Sun, 22 Dec 2024 07:12:17 -0800
Zachary Snow ***@***.***> wrote:

> Thank you for filing this issue! For these localparams, sv2v relies on the sign and size extension of assignment like contexts defined in the Verilog standard to simplify its conversion. That is, [...]

Well, the problem is that verilator complains about it and since
comments are elided I can't turn the warnings off as verilator reads
comments to know that an issue has been spotted and deemed nondangerous.
So sv2v is generating hundreds of verilator warning messages that come
from the conversion alone, making them useless for diagnostics, and
hiding any actual problems.

I'm afraid one has to do this. The behavioral semantics of code is
not quite the full semantics.

Is there difficulty in using the same "[3:0]" type that is output to
generate "4'b" instead of "1's" ?

It looks like practically a one-line change in the haskell.

best regards

PTB

zachjs, 2025-04-06:
> Is there difficulty in using the same "[3:0]" type that is output to generate "4'b" instead of "1's" ? It looks like practically a one-line change in the haskell.

Indeed, commenting out the following logic in `src/Convert/UnbasedUnsized.hs` should yield the behavior you desire.
```Haskell
convertAsgn (lhs, UU bit) =
    (lhs, literalFor bit)
```

It is technically possible to add a command line flag to toggle this behavior, but I fear this is but one of the many arbitrary decisions a linter could make. verilator itself has the option to disable this warning, but I'm guessing that's not what you want. Before proceeding, I'm at least curious if you could try that local sv2v modification above and see if it actually yields the overall behavior you want.

If the complaints are isolated to the `sv2v_uu` localparam declarations, it may be possible to use `sed` to post-process the output to convince verilator we know what we're doing for these particular snippets.