zachjs/sv2v #302: doesn't deal with for loops with multiple initializations

pbreuer, 2024-12-22:
sv2v doesn't seem to like this example from https://www.chipverify.com/systemverilog/systemverilog-for-loop or similar: 

```
               for (int i = 0, j = 2; i < $size(array); i++) begin
			array[i][j] = "0";
                        $display ("array[%0d] = %s, %0dth index replaced by 0", i, array[i], j);
		end
```
I'm not trying exactly that, but that's the spirit and it is what I was following for format.
 It seems to be objecting to the  comma, not the type declaration, and it happens both without the type declaration and with it. I don't think sv2v handles inline declarations there anyway?

Multiple increments are also legal. Is this a "which standard" question? That page claims "There can be multiple initializations done in the first part of a for loop" (as in C).

The error from sv2v is just a parse pointer: "Parse error: unexpected token ',' (Sym_comma)"

Regards

PTB

zachjs, 2025-02-17:
Thank you for filing this issue! I definitely _intend_ to support this syntax:
https://github.com/zachjs/sv2v/blob/aa0a885699ca8f4fc1d07ef34fb041c6f37b1ba3/src/Convert/ForAsgn.hs#L4-L6
so I suspect I'm missing something. In fact, both "multiple initializations" and "multiple increments" are supported.

For example, when I convert:
```SystemVerilog
module top;
    initial
        for (int i = 0, j = 2; i < 5; j++, i++)
            $display("i=%0d j=%0d", i, j);
endmodule
```
through sv2v, I get:
```Verilog
module top;
	function automatic signed [31:0] sv2v_cast_32_signed;
		input reg signed [31:0] inp;
		sv2v_cast_32_signed = inp;
	endfunction
	initial begin : sv2v_autoblock_1
		reg signed [31:0] i;
		reg signed [31:0] j;
		i = 0;
		for (j = 2; i < 5; {j, i} = {sv2v_cast_32_signed(j + 1), sv2v_cast_32_signed(i + 1)})
			$display("i=%0d j=%0d", i, j);
	end
endmodule
```
When I run that output through iverilog, I get:
```
i=0 j=2
i=1 j=3
i=2 j=4
i=3 j=5
i=4 j=6
```
For reference, I'm using sv2v `v0.0.12-27-g4ec99fc` and iverilog `13.0 (devel) (s20221226-513-gef7f0a8f3)`.

Is it possible that the parse error is caused by a different part of the input? (Perhaps the source location sv2v gives is bad?) What `sv2v --version` are you using?

zachjs, 2025-04-06:
@pbreuer Did you have the chance to review my questions above? As I mentioned, I believe this feature is already supported, including the specific snippet included in your issue description. I'm eager to help if you can share more information!

pbreuer, 2025-04-06:
Sorry but I'm not using s2v at the moment

I should have been more explicit with the test case!

closed 2025-04-06 by zachjs