zachjs/sv2v #304: Parameter redefinition circular reference

alexforencich, 2025-02-01:
I am running in to an issue where sv2v is creating circular references with parameters.  I don't  currently have a minimal example to share, but sv2v is generating code like this:

				localparam DATA_W = _param_F4CDB_DATA_W;
				localparam CTRL_W = _param_F4CDB_CTRL_W;
				// snip some more parameters
				// snip a bunch of wires
				localparam KEEP_W = _param_6FA32_KEEP_W;
				localparam TX_USER_W = _param_6FA32_USER_W;
				localparam RX_USER_W = _param_A2E34_USER_W;
				localparam MAC_CTRL_EN = 1'd0;
				localparam TX_USER_W_INT = (MAC_CTRL_EN ? (PTP_TS_EN ? (TX_PTP_TAG_EN ? TX_PTP_TAG_W : 0) + 1 : 0) + 1 : TX_USER_W);
				localparam _param_6FA32_DATA_W = DATA_W;
				localparam _param_6FA32_KEEP_W = KEEP_W;
				localparam _param_6FA32_USER_W = TX_USER_W_INT;

This causes Quartus to complain:

    Error (10234): Verilog HDL Defparam Statement error at fpga.v(1467): parameter "_param_6FA32_KEEP_W" cannot be assigned to its own value -- can't resolve self-referential loop File: /home/alex/Projects/.../fpga.v Line: 1467
    Error (10234): Verilog HDL Defparam Statement error at fpga.v(1461): parameter "KEEP_W" cannot be assigned to its own value -- can't resolve self-referential loop File: /home/alex/Projects/.../fpga.v Line: 1461

zachjs, 2025-02-01:
I agree that output looks suspect! A reproducer would be very helpful to isolate the issue. Thank you!

alexforencich, 2025-02-01:
This test code:

    interface test_if #(
        parameter DATA_W = 8
    )
    ();
        logic [DATA_W-1:0] tdata;

        modport src (
            output tdata
        );

        modport snk (
            input  tdata
        );

    endinterface

    module mod1 (
        test_if.snk in
    );

    localparam DATA_W = in.DATA_W;

    endmodule

    module mod2 (
        test_if.snk in
    );

    localparam DATA_W = in.DATA_W;

    test_if #(.DATA_W(DATA_W)) t();

    mod1
    mod1_inst (
        .in(t)
    );

    endmodule

    module top();

    test_if #(.DATA_W(16)) t();

    mod2
    mod2_inst (
        .in(t)
    );

    endmodule

produces:

    module top;
	    localparam _param_B1C85_DATA_W = 16;
	    generate
		    if (1) begin : t
			    localparam DATA_W = _param_B1C85_DATA_W;
			    wire [15:0] tdata;
		    end
		    if (1) begin : mod2_inst
			    localparam DATA_W = _param_B1C85_DATA_W;
			    localparam _param_B1C85_DATA_W = DATA_W;
			    if (1) begin : t
				    localparam DATA_W = _param_B1C85_DATA_W;
				    wire [15:0] tdata;
			    end
			    if (1) begin : mod1_inst
				    localparam DATA_W = _param_B1C85_DATA_W;
			    end
		    end
	    endgenerate
    endmodule

zachjs, 2025-02-17:
Thank you very much filing this issue and sharing such a clean and simple reproducer! Your effort made my work very easy.

sv2v was indeed needlessly prone to generating colliding parameter names when converting interfaces. I've just pushed 4ec99fcffd2c533c4a2c74ad6f56278f2672f05d, which appears to resolve the issue. Please let me know if it works for you!

alexforencich, 2025-02-18:
I think that might solve it; but there are a couple of other issues that need to be fixed before I can actually test this out properly.  I will report back here as soon as I know for sure.

zachjs, 2025-04-06:
@alexforencich Thank you again for sharing that reproducer! It was very helpful. As I believe the specific bug you encountered in this thread is now resolved, I am closing this issue. Please feel free to reopen if necessary!

closed 2025-04-06 by zachjs