zachjs/sv2v #308: Some tools have issues with hierarchical references

alexforencich, 2025-02-18:
When using SV interfaces, sv2v likes to produce lots of hierarchical references.  Makes sense as it keeps things similar to the original code.  However, not all tools fully support hierarchical references.

Quartus seems to have issues with passing hierarchical references to functions, complaining that it `can't pass value from actual to argument <x>`.  A workaround seems to be to define a shadow signal for the signal in question.

ISE seems to not support hierarchical references at all, or at least it has some issues with the implementation.  It returns errors like `<signal> is not declared under prefix <entity>`.

zachjs, 2025-02-23:
> When using SV interfaces, sv2v likes to produce lots of hierarchical references. Makes sense as it keeps things similar to the original code.

The motivating factor was really ease of conversion.

> simply provide an option to avoid generating hierarchical references entirely.

I don't think this is nearly as easy at it sounds. If it were possible it at all, it would likely be a separate "post-processing" conversion phase. Can declarations within a deeply-nested generate block be referenced elsewhere so long as they have a unique name?

Perhaps narrowing down what sorts of hierarchical references given them trouble would make this issue more tractable. "Function arguments" might be narrow enough, but "everywhere" may not be tractable. Does ISE support generate loops and references to declarations within them?

alexforencich, 2025-02-23:
The quartus issue definitely smells like a quartus bug which could potentially be fixed by Altera/Intel, but since ISE is no longer supported, we'll have to figure out how to work around the limitations of the latest available release.  Also, one additional problem with ISE is that it uses different synthesis engines depending on the target device, and the different engines have different limitations.  Main thing I am aware of offhand is that $clog2 is supported on 6-series, but not on 4/5 series, even in the same release of ISE.  I'll look in to running a few tests targeting maybe Spartan 6, Spartan 3, Virtex 5, and Virtex 4.  I haven't really used hierarchical references much myself; I suspect there could be larger issues in how they're handled by some synthesis tools, especially older ones.

zachjs, 2025-04-06:
As mentioned in https://github.com/zachjs/sv2v/issues/307#issuecomment-2677097208, I'm very curious if something like Yosys could help to further simplify your Verilog. Please let me know what you find! sv2v was originally built to target the (limited!) feature set of Yosys, so if you run into trouble combining these two tools, there will likely be significant interest in tracking it down. Although we'd ideally support the "lowest common denominator" toolchain, unsupported software downstream won't do us any favors here.