zachjs/sv2v #309: ISE has issues with $clog2

alexforencich, 2025-02-18:
ISE has issues with code containing `$clog2`.  For 6-series, it is not allowed in `localparam`, only `parameter`.  For parts older than 6-series, it is not supported at all.  Perhaps sv2v can define a `clog2` macro and substitute that for uses of `$clog2`, perhaps this can be configurable via a command line option or similar.

zachjs, 2025-02-23:
`$clog2` was added to Verilog in IEEE 1364-2005, released nearly 19 years in April 2006. There's certainly no reason to support it in some constant expressions but not others. Would Xilinx/AMD be receptive to a bug report in this case?

Perhaps we could have a flag to replace all `$clog2` with the following constant function. Can you try this substitution in your code and confirm whether this actually works in ISE? (I don't know what other restrictions it might impose.)
```Verilog
function automatic integer clog2;
    input integer inp;
    clog2 = 0;
    for (inp = inp - 1; inp > 0; inp = inp >> 1)
        clog2 = clog2 + 1;
endfunction
```

alexforencich, 2025-02-23:
I wish, but ISE is currently no longer supported, so we're stuck with what we've got.  I'll look into trying that out and report back.  Hopefully that works, a macro is also a potential option, but would be decidedly less clean.

zachjs, 2025-04-06:
@alexforencich Did you have any luck with the `$clog2` substitute above?