zachjs/sv2v #310: Fail to recognize the sequence.

ForJill, 2025-02-24:
The mistake shot:
[image]

Relevant code:
[image]

de0u, 2025-02-24:
When possible, it's best to submit error messages and code sequences as text rather than as images.  Among other reason, text can easily be searched, but a picture of text is much harder to index.

In GitHub it is possible to quote error messages and code blocks using standard Markdown, for example starting a code block with a line containing three back-ticks (\`\`\`) and ending the block with another line like that.

Since the error message complains about ``sequence`` being unexpected, it is possible that what came before the code block was part of the issue.  Is it possible to submit a complete-but-small System Verilog example, containing this sequence or a similar one, that results in this error message?

zachjs, 2025-04-06:
Bumping this thread, @ForJill. As @de0u explained, it would be very helpful to have a complete example to help me reproduce and address this issue.

zachjs, 2025-05-18:
The following snippet passes through sv2v cleanly. (The assertion is removed, unless `-Eassert` is provided.) This confirms the parsing error has to do with whatever _precedes_ the code in the screenshot, which hasn't been shared. Without further input from @ForJill, I'm not sure there's anything I can do here. For now, I'm going to close this reflecting my inability to reproduce it with the information available. @ForJill, please feel free to comment or reopen whenever you get the chance!

```SystemVerilog
module top;
    sequence FullHandshake_S;
        $rose(prim_alert_rxtx_fpv.alert_tx_out.alert_p) ##1
        $rose(prim_alert_rxtx_fpv.alert_rx_out.ack_p) &&
        $stable(prim_alert_rxtx_fpv.alert_tx_out.alert_p) ##1
        $fell(prim_alert_rxtx_fpv.alert_tx_out.alert_p) &&
        $stable(prim_alert_rxtx_fpv.alert_rx_out.ack_p) ##1
        $fell(prim_alert_rxtx_fpv.alert_rx_out.ack_p) &&
        $stable(prim_alert_rxtx_fpv.alert_tx_out.alert_p)
    endsequence
endmodule
```

closed 2025-05-18 by zachjs