zachjs/sv2v #312: Request for a Release

sifferman, 2025-03-19:
Hello. Could there be a release of the latest build?

A few of my students have needed to build sv2v from source to grab some of its latest features. alexforencich has mentioned the same:

> Quartus Prime Lite/Standard does not support `$fatal`, even in SV mode.  And apparently this fix hasn't made it into a release yet, but when I build it from source then this issue has been solved. 

 _Originally posted by alexforencich in [#273](https://github.com/zachjs/sv2v/issues/273#issuecomment-2628780042)_

Thanks!

zachjs, 2025-03-20:
I have created a new release at https://github.com/zachjs/sv2v/releases/tag/v0.0.13 and https://hackage.haskell.org/package/sv2v-0.0.13. Please let me know if this looks good to you!

sifferman, 2025-03-20:
Thanks! Looks perfect!

closed 2025-03-20 by sifferman