zachjs/sv2v #313: [Feature Request] Support `+define+foo=bar+fizz=buzz` on command line

infinitymdm, 2025-04-04:
The `-Dfoo=bar` syntax is great, and aligns well with tools like cmake, but doesn't match up with the industry standard `+define+foo=bar` syntax. `+define` also typically allows multiple defines in one long string.

zachjs, 2025-04-06:
I'm a little hesitant to add another way to define macros. The current `-D` interface matches that of iverilog and the open source Yosys Verilog frontend; verilator supports both.

If there is interest, I would happily consider a pull request, though I think it may be a bit tricky to do with the command line parser that sv2v uses.

As a workaround, you could consider converting such strings before passing them to sv2v, e.g., `sed -E 's/\+define//' | sed -E 's/\+/ -D/g'`.

infinitymdm, 2025-04-07:
If I ever decide to dabble in haskell you can expect a PR. At present though, I'll certainly make use of that workaround. Thanks!

zachjs, 2025-05-18:
I'm glad to hear that workaround works for you! I think that's still my preferred solution, so I'm going to close this issue as completed. Please feel free to comment or reopen as appropriate!

closed 2025-05-18 by zachjs