zachjs/sv2v #317: Consider relaxing upper bound for 'base' dependency to support newer GHCs

Emin017, 2025-05-16:
Hi sv2v maintainers,

When packaging sv2v for Nixpkgs (a Linux distribution and package manager), we've encountered a build issue due to the current upper bound on the `base` dependency (specified `resolver: lts-22.19` in stack.yaml). This prevents building sv2v with GHC versions that use `base-4.19.*` or newer (e.g., GHC 9.8 and later).

We've found that sv2v builds and seems to function correctly when this bound is relaxed (we used a "jailbreak" method in Nixpkgs to bypass the restriction).

Could you please consider relaxing the upper bound on `base` in your `.cabal` file to allow compatibility with newer GHC versions? For example, something like `base < 4.21` (or an appropriate higher version you are comfortable supporting after testing) would be very helpful.

This would simplify packaging for distributions and ensure broader compatibility with the Haskell ecosystem.

Thank you for considering this!

zachjs, 2025-05-18:
Thanks for the heads up! I've taken some steps to improve my versioning posture on https://github.com/zachjs/sv2v/tree/version. Can you see if https://hackage.haskell.org/package/sv2v-0.0.13/candidate works in your nix builds? (I'll use a different version number if I make the candidate official.)

I noticed you asked for `base < 4.21` rather than `base < 4.20`. Do other Haskell package maintainers (who use Stack) typically use `--resolver nightly` when uploading their packages?

Emin017, 2025-05-19:
> Thanks for the heads up! I've taken some steps to improve my versioning posture on https://github.com/zachjs/sv2v/tree/version. Can you see if https://hackage.haskell.org/package/sv2v-0.0.13/candidate works in your nix builds? (I'll use a different version number if I make the candidate official.)

Not quite sure how to test https://hackage.haskell.org/package/sv2v-0.0.13/candidate, the sv2v packages in nixpkgs should be automatically imported from hackage (It used to be broken on nix build). But it seems that the https://github.com/zachjs/sv2v/tree/version works on my nix builds.

> I noticed you asked for `base < 4.21` rather than `base < 4.20`. Do other Haskell package maintainers (who use Stack) typically use `--resolver nightly` when uploading their packages?

Apologize for my typo, I think the latest lts version is enough for packaging.

zachjs, 2025-06-23:
I just published https://hackage.haskell.org/package/sv2v-0.0.13.1. Please let me know if this fixes the main nix build!

Emin017, 2025-06-24:
> I just published https://hackage.haskell.org/package/sv2v-0.0.13.1. Please let me know if this fixes the main nix build!

I tested it on my machine, and version 0.0.13.1 can complete the build without going through the `doJailbreak` configuration. Thank you for the modification!

closed 2025-06-24 by Emin017